OpenEVSE/openevse_esp32_firmware #557: RAPI command GT get time (RTC) not working

cecil-t, 2023-03-06:
When I issue RAPI Command: $GT
I get the response: $OK 165 165 165 165 165 85^1F
Instead of the correct: yr mo day hr min sec  (2 digits each)
The response seems to be consistently $OK 165 165 165 165 165 85^1F regardless of time.

[image]

jeremypoulter, 2023-03-06:
Do you have the RTC enabled LCD module? The RTC has not been included for a while as the ESP will normally get the time via NTP

cecil-t, 2023-03-06:
I guess not, this is a brand new unit.  I'm trying to plug this into my existing ecosystem with other units and software, but this time function is the only thing that appears to not be working.  Can the $GT function return the unit's date/time regardless of where it was set from?

jeremypoulter, 2023-03-06:
You need to use the HTTP API not RAPI. You can get the current time from the `/status` endpoint.

cecil-t, 2023-03-06:
I would have to rebuild my entire app that was designed years ago to move to the HTTP API.  I was hoping this was an easier fix, since time seems to be the only thing not working in the RAPI.  And then also I'm not sure my older devices with older Wi-Fi and controllers fully support the HTTP API.

jeremypoulter, 2023-03-06:
Sorry we are deprecating RAPI so can't help much more than to recommend moving to the HTTP API, also RAPI is just a pass through to the EVSE so would more be an issue for the EVSE firmware. 

The v4 firmware is supported on any ESP32. Older ESP8266 devices could be upgraded with the Adafruit ESP32 Huzzah. It is a drop in replacement for the ESP8266 equivalent.

cecil-t, 2023-03-06:
Understood, no worries, it's on my list one day.  Thanks for the info on the ESP8266 devices.

closed 2023-03-06 by cecil-t