OpenEVSE/openevse_esp32_firmware #558: Corrupted flash after update ?

KipK, 2023-03-06:
Seems with current dev branch, the configuration reset itself sometimes after an update. Don't know yet if it's corrupted or not. 
Some configuration data stays like Wifi settings but some others don't. 

I've implemented the backup/restore on the UI side but that would be great to find what's happening here.

KipK, 2023-03-06:
I see in the config after an update failing the conf, that most settings like mqtt has been reseted to default values.
There's also the "factory_write_lock" set to true.
The only recent change touching the config seems to be related to this factory_lock thing.

KipK, 2023-03-06:
If I can reproduce with a standalone esp32 probably yes.

KipK, 2023-03-09:
Nothing different on the serial console when config fails.
I can reproduce the problem by flashing twice the same fw, first time it keep the config, second time not.

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[   164][E][vfs_api.cpp:105] open(): /littlefs/schedule.json does not exist, no permits for creation
$GV^35
`

the vfs_api log always happend here on my test esp32 not plugued to openevse module, don't know if it's the case on a real situation, but doesn't seems to caus eproblem to schedule anyways

jeremypoulter, 2023-03-09:
I think that is the RAPI serial port, there is one for debug output that should give more info. What board are you using?

KipK, 2023-03-10:
ah you're right I was on the usb port on an esp32 dev wroom.

KipK, 2023-03-10:
here is the log:

`Restarting...
                                                                 
NetManagerTask::wifiStop 1
                                                     
                                                                        
OpenEVSE WiFI 770c
                                                             
Firmware: 4.1.8
                                                                
Build date: Mar 10 2023 12:42:22
                                               
IDF version: v4.4.3
                                                            
Free: 253064
                                                                   
Got 65535 ÿ from EEPROM @ 0x0c00
                                               
No JSON config found
                                                           
Got 139 { from EEPROM @ 0x0000
                                                 
Found stored JSON {"ws":"xxxxxxxxx","wp":"xxxxxxxxxx","lan":"fr","hn":"openev
se2","tz":"Europe/Paris|CET-1CEST,M3.5.0,M10.5.0/3","ldt":"none","f":2097160}
  
ssid changed
                                                                   
pass changed
                                                                   
lang changed
                                                                   
hostname changed
                                                               
time_zone changed
                                                              
limit_default_type changed
                                                     
factory_write_lock changed
                                                     
Saving config
                                                                  
`

KipK, 2023-03-10:
one of the latests commits seems  to have solved the problem, I close the issue for now as I can't reproduce anymore.

KipK, 2023-03-15:
Reopening, it happend again, and worst, nows reset config after 2 reboots.

Reproduce: reset twice, then refresh UI  ( note for later: I have to fix this refresh needs on the UI )

commit 5155bb1e (2023-03-23): fix config beeing erased at reboot introduced by Limit
fix #558
commit abfb985c (2023-03-23): Merge pull request #576 from KipK/master

fix #558 config erased at reboot
closed 2023-03-23 by jeremypoulter