OpenEVSE/openevse_esp32_firmware #560: Scheduler wrong calculation of current event

KipK, 2023-03-08:
Related to discussions here for more details: https://github.com/KipK/openevse-gui-v2/issues/31

How to reproduce:
Set a timer to Active few minutes before current time , with same current hour.
Set a second timer to Disabled that should happend later in the day.

It will set the state of evse to disabled instead of using first timer.

KipK, 2023-03-10:
It looks like it's related to Time Zone.
On my case the timers start 2 hours later than planned ( I'm on GMT +2 )

fhteagle, 2023-03-10:
I think you are correct that it is a TZ problem in the scheduler calculation. I tested both 2 and 3 timer schedules, and was able to affect the results by moving an active line across a time zone gap. I.e. I am in UTC-7 at the moment. If the active was less than now +7 hours, it was incorrectly applied and activated the unit. If the active was more than now +7 hours, it was correctly disregarded for the calculation.

KipK, 2023-03-10:
I can change the UI to send utc time for /schedule endpoint and parse time back to local time, but is it what's the api expect @jeremypoulter ? looks to be a bit boilerplaty but perhaps there's a reason to not use local timer in scheduler ?

KipK, 2023-03-10:
I remember I have changed the time handling in the scheduler part of ui few days ago, I bet that's what introduced the bug and it is UI bug only,  it was working ok before. :)

KipK, 2023-03-10:
I have checked in scheduler.cpp and 
localtime_r(&utc_time.tv_sec, &local_time)

reports 14h20 instead of 15h20 here, ( i'm UTC+1 in winter) 

but reading the code seems to me it expects local time in timers so we ahve a problem here

jeremypoulter, 2023-03-10:
:( arrrr timezones.... Will take a look

jeremypoulter, 2023-03-11:
I fixes a few time zone related issues in #569, can you check?

KipK, 2023-03-11:
still same problem.

KipK, 2023-03-11:
have set a timer at 19:02 to enable,  and disable at 6am, it is 19h11, state is still disabled

KipK, 2023-03-11:
interesting, if i set time to manual it then works. But manual has no timezone  isn't, it ?

KipK, 2023-03-11:
oups forget it, i haven't test with your PR, thought you've updated the master :)

KipK, 2023-03-11:
fixed by #569

fhteagle, 2023-03-17:
The dev build from yesterday still has this bug apparently. Is there a different build I need to be testing to verify this fix?

KipK, 2023-03-22:
@fhteagle you can try my build from today it has all the fixes waiting to me merged

https://drive.google.com/open?id=1FZYEWYd9MTqPxts0MquPUQlYGsO7DoNm&authuser=hotlobster75%40gmail.com&usp=drive_fs

fhteagle, 2023-03-22:
@KipK  -

Scheduler seems to be working better in this version, yes. I was not able to induce any of the scheduling failures that we have been seeing.

closed 2023-03-30 by glynhudson