OpenEVSE/openevse_esp32_firmware #573: HTTP API accepts "charge_current" below 6A, then GUI displays wrong actual offered current

fhteagle, 2023-03-17:
If one issues

 curl --request POST \
  --url http://openevse.local/override \
  --header 'Content-Type: application/json' \
  --data '{
  "state": "active", "charge_current": "4"
}'

The unit responds:
{"msg":"Created"}

Then GUI shows manual override, with 4A on the slider. However, 6A is actually offered to the vehicle, which AFAIK is in line with the lowest value available in the J1772 / J3068 specifications. 

Should the HTTP API offer a failure, or "we fixed your bad command" response message instead?

Either way, the GUI should not be showing an inaccurate number of amps compared to what is actually being offered to the vehicle at this time.

jeremypoulter, 2023-03-17:
Hum... That is a good question... I would be tempted to say no. 

In general the ok is just an indication that the underlying claim has been set, and these are requests to set to value, the override has more weight but can still be changed, eg the current shaper, or temperature monitoring. That being said the minimum 6A is an exception as this cannot be changed (and in practice the hardware max) but would treating them as a special care cause confusion?

fhteagle, 2023-03-17:
The behavior of the hardware and the results of an HTTP API call after a bad charge_current value is POSTED seem valid in my opinion. It is just the GUI behavior for the low end that needs to be changed. The HTTP API response indicating "we accepted your command but adjusted it" would be nice, but not strictly necessary.

KipK, 2023-03-22:
If the API gives a less than 6amp charge_current in /claims or /override it will display it with this value. 
I ça make an exception but wouldn't it be better to have the api correcting the value itself?

KipK, 2023-03-23:
fixed on UI by https://github.com/KipK/openevse-gui-v2/commit/44bea011809f389b782d66d648d5920e515453c4

closed 2023-03-30 by glynhudson