OpenEVSE/openevse_esp32_firmware #584: Auto version number not added when built by GitHub

jeremypoulter, 2023-03-30:
When the firmware is built by the GitHub workflow no version information is added to the build with an error `fatal: No names found, cannot describe anything.`

This looks to be because GitHub checks out with depth=1, https://stackoverflow.com/questions/66349002/get-latest-tag-git-describe-tags-when-repo-is-cloned-with-depth-1

I think we may need to update to use the `GITHUB_REF_NAME` env var, if that is set, and is start with `v` then use as is, if it does not start with `v` then append the git hash (`git rev-parse HEAD`). If `GITHUB_REF_NAME is not set then`git describe --tags` with a fallback to `git rev-parse HEAD`.

KipK, 2023-03-30:
Looks good yes, and as told you in pm, we need a field in the github release endpoint api having the version number :
https://api.github.com/repos/OpenEVSE/ESP32_WiFi_V4.x/releases

could be "assets" objects "label" property ( should be the same for each binary ) 
or directly in the tag_name but it will messup the release/build page with different tags for each dev build

KipK, 2023-03-30:
Here is  how it looks with daily builds:

[image]

I need to match the Installed field ( taking the build version from the fw ) with the Latest one ( took from github api  , for now it takes "name" field , this should be changed.

PR #589 (2023-04-01): Fixes to the auto version number script
Updated the auto version number script to use the correct GitHub environment variables to get the version number when running on GitHub Actions.

Added the `_modified` suffix to the version number when the working directory is not clean. This is to ensure that the version number is always different when the working directory is not clean.

Added the `GIT_HASH` define with the short hash of the current commit. This is to make it easier to identify the exact commit that was used to build the binary.

Fixes #584
commit 551a1e89 (2023-04-01): Fixes to the auto version number script

Updated the auto version number script to use the correct GitHub environment variables to get the version number when running on GitHub Actions.

Added the `_modified` suffix to the version number when the working directory is not clean. This is to ensure that the version number is always different when the working directory is not clean.

Added the `GIT_HASH` define with the short hash of the current commit. This is to make it easier to identify the exact commit that was used to build the binary.

Fixes #584
closed 2023-04-01 by jeremypoulter