OpenEVSE/openevse_esp32_firmware #586: MQTT using TLS v1, depricated by Mosquitto

rbray89, 2023-03-31:
If possible, TLSv1.3 should be used in the MQTT implementation. I have not investigated if the mongoose library used supports this or not.

rbray89, 2023-03-31:
Hmmm... I may have the wrong idea here... 

I was assuming the TLS version was soff, but it seems the actual issue may be that the MQTT service doesn't have the letsencrypt CAs available to verify signing.

jeremypoulter, 2023-03-31:
The root CAs are defined in <https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/src/root_ca.cpp> so should have the Lets Encrypt CA, it is used by EmonCMS, I can well believe mongoose is using TLS v1, I would have to check, but can you detail your issue?

rbray89, 2023-04-01:
Interesting. 

I setup a standard Mosquitto install using Lets Encrypt issued certificates.

When setting OpenEVSE MQTT up, I had to disable self-signed cert rejection in order to connect.

Additionally, it may have been rejecting my wildcard cert outright, but I'd have to roll things back to be sure.

jeremypoulter, 2023-04-01:
What is the main subject of the certificate and what are the MQTT connection details?

rbray89, 2023-04-04:
@jeremypoulter Ah, you hit the nail on the head there.

Seems like openEVSE MQTT doesn't respect SAN entries. Using the CN from the Subject did work though.

Not a major issue as they both resolve to my site, but should either be documented or changed to allow SAN entry verification.

rbray89, 2023-04-04:
Hmm... So I *thought* I had updated my cert to have my sub-domain as a SAN entry, but it still has the wildcard subdomain entry.

However, that should be valid as well anyways. So I thin the point still stands, I just don't know if I'd be seeing this issue if my cert included the non-wildcard subdomain in the SAN list.

closed 2026-09-21 by github-actions[bot]