OpenEVSE/openevse_esp32_firmware #587: Feature Request: Minimum Idle Time with reset for relay wear reduction and avoiding power peaks

pdhoogh, 2023-03-31:
I am re-iterating this as requested by @jeremypoulter and agreeing with @soloam.

The intention is to turn off the charging (> Disable) as soon as the "available current" drops below 6 amps. A timer is started at the same time. The timer is reset every time the available current is calculated to be below 6 amps. When the timer runs out, the charge is resumed.

This will avoid import power peaks, because the EV charging is turned off as soon as the import power peak is detected. It will also save the relay in the OpenEVSE, as with this feature, OpenEVSE will wait for stable power availability before it resumes charging. A number of home appliances are cycling power with short intervals:

- Laundry Dryers
- Ironing stations
- Microwave ovens that run at "reduced power setting"
- Washing machines
- Dishwashers
- Solar PV installation with half open skies
- more I forgot to add...

I am repeating @solam's example below:

Let's assume a idle time of 10 minutes.

10:00 - Goes bellow 6 - System goes idle
10:30 - Remains bellow 6 - Remains Idle
10:35 - Goes to 10 - Remains idle for 10 min
10:40 - Goes bellow 6 - Remains idle and reset timer
10:50 - Goes to 10 - Remains idle for 10 min
11:00 - Remains over 6 - Resumes Charging

jeremypoulter, 2023-03-31:
That is what the smoothing attack and decay values do.

If you want to prioritise being off then set a large decay value and a small attack value (the defaults are the other way round), this will mean it takes longer to switch on and quicker to switch off. When we were developing the algorithm we did try an idle delay but it mostly just waisted PV.

Even the minimum active timer is of questionable value.

pdhoogh, 2023-03-31:
@jeremypoulter your reaction surprises me. I seem to remember you were in agreement at an earlier time to have the debouncing timer as a "Minimum time OFF" timer rather than a "Minimum time ON" timer. And I specifically remember you made the comment you did not know why the choice was made for the latter.

I agree it may "waste" PV power. But the "Minimum time ON" can help create huge import power peaks. I have seen this many times. When a 4kW load (like my swimspa) kicks in within the 10 minutes the OpenEVSE decided to start charging the EV, the import power peaks to 6 kW. Anything over 2.5 kW sustained import power is costing me on the utility bill. I do not mind "wasting" PV, that is not so costly.

Using a small attack value won't stop the OpenEVSE from drawing at least 1400 Watt if the "Minimum Time ON" timer has not ran out.

So I am noting you do not feel like implementing this feature. No problem. I'll keep my Python code running, disabling and enabling the charger using MQTT. As with many similar things I encountered in life, time is my friend.

jeremypoulter, 2023-03-31:
I have never been a fan of the timers, the smoothing produces much better results in a lot more cases. For the use case you described, waiting for a stable current, the smoothing is exactly what you want. The on timer can be configured to a low value. 

That being said I wouldn't reject a PR that implemented a minimum off timer.

soloam, 2023-03-31:
In this case stable current is not very useful. The relay only turns off under 6 amp, everything above that is ok to charge the car... What I don't want is my relay going on/off several times just because my washing machine or dryer are making my power go up and down. but if they drain the power and my evcharger is over 6 amps, no problem.

pdhoogh, 2023-03-31:
Agreed with @soloam. Stable current is not the issue. Stable availability above 6 Amps is what counts. My EV has no issue following frequent pilot signal changes.

jeremypoulter, 2023-03-31:
Sorry that is what I meant, the smoothing filters out any noise/spikes in the available current, by turning down the attack the EVSE will be much slower to turn on and likewise turning up the decay makes it quicker to turn off.

pdhoogh, 2023-03-31:
OK, I'll try out your suggestion. What is the unit used for the filter value? If it is 1.0, does that mean a 1 minute exponential low pass filter ?

pdhoogh, 2023-03-31:
This is indeed exponential smoothing:
[image]

[image]

Using the alpha as the variable to set is a bit confusing for thinking in the time domain.

With a period (update rate) of 10 seconds, a Attack / Decay value of 0.1 is like an exponential filter with a time constant of 1.66 minutes. A time constant of 10 minutes requires an Attack Value of about 0.016. That is what I will use for testing.

[image]

Below is a quick reference chart of Attack Value vs. Time Constant (exponential filter) assuming a 10 second sample time:
[image]

jeremypoulter, 2023-03-31:
Yeah, adding validations to a lot of the config is on the to do list, but I have not written the issue yet. As an FYI here are some of the original simulations we did, using real data, https://docs.google.com/spreadsheets/d/1GQEAQ5QNvNuShEsUdcrNsFC12U3pQfcD_NetoIfDoko/edit?usp=sharing, IIRC these are fairly close to what is implemented in the code. At some point I want to fix up the divert simulator again so people can use it to test on their own data...

pdhoogh, 2023-04-01:
### Input Validation Options
**The better option** is that the divert code itself does an initialization / data validation routine before doing anything else. That has been a best practice in process control for decades now. And the first thing I coded in my control programs.
**The next best thing** is to do data validation in the GUI. And prevent input outside data validation limits.
**The minimum thing to do** is to put a message in the GUI that the value to be entered should be within the given limits.

@KipK may consider the latter two for the new GUI. The limits for Attack / Decay in the filter equation used are 0 and 1. 1 means essentially no filter. But 0 does not make a lot of sense, a time constant of infinity, that is essentially a killer and should be treated in the divert code as "no filter". If the divert code does not do that, than another practical low limit should be used, e.g. 0.001. 

### The Best Option
In my opinion **the  best** option is to use time constants as input parameters rather than filter factors. I do not think many people understand filter factors very well. And the effect of the filter in the time domain depends on the sample rate of the metering data. Depending on how reliably the data stream from the metering is coming through, the filter will behave differently in the time domain. We assume a 10 seconds sample time, but with data not coming through in time, the behavior in the time domain may become an absolute mess. So the GUI should ask for a time constant in minutes and the divert code should calculate the filter factor every pass depending on the update period it sees. 
[image]
Where Delta T is the measured sample period in minutes and Tau is the filter factor in minutes. 

Here too, data validation should be applied, where zero should be seen by the divert code as "no filter" and negative values are not allowed. The advantage of this approach is a more stable behavior in the time domain and the upper limit is not of much consequence. Also there is less need for a simulator.

### Original Simulations
Indeed, the way the filters are set up are the best approach in the old realm where utility companies were not worried too much about the dent PV owners put in the network. But as more and more people install PV, also companies doing that on an industrial scale, evening out these bigger and bigger dents is becoming a serious nightmare for the utility companies. So the situation has changed. They react to this by having new legislation and tariffs. And we need to react to those changes following the tariff structure imposed on us.

The tariffs are primarily intended to safeguard the profit for the companies involved, not necessarily what is best for the environment. Capitalism does not care about the environment. So our controls need to counter the tariffs, even if they do not seem to be environmentally sound. To make sure we are not the ones ending up paying for the environment as well as the company profits. And force the tariffs to get closer to what is environmentally sound. Those tariffs will become environmentally sound only when technology and the cost of it will enable customers to go off-grid. Total loss of business will make them think twice.

This ideological piece was needed to clarify where I and @soloam are coming from. And why we need to change the way OpenEVSE behaves. I.e. if grid import power gets positive over a certain set value, it may be even when it becomes higher than zero, the EV charging should stop _immediately_. Even if this means "wasting" solar PV power by injecting it into the grid afterwards while OpenEVSE is waiting for "stable" power as @jeremypoulter puts it. Because the import peaks cost more than the "wasting" of PV power.

KipK, 2023-04-01:
I've changed the UI using slider on 0.01 - 1.0 range
Do we really need going under 0.01? ( 1%)

[image]

pdhoogh, 2023-04-01:
Great!

With a 10 second period, a filter factor of 0.01 is about a 16 minute time constant. In my opinion that is a good practical limit.

soloam, 2023-04-01:
I think that the main issue here is the use case that someone is giving to the Current Shaper... For me the shaper is to ensure that I can always supply my car with the maximum current available without having a my energy supplier cutting the supply for over usage. On my Home my maximum current is 32 Amp, my car is charging at 20 Amp maximum! This allows the rest of the house to use 12 Amp. If I turn on washing machine and dishwasher that value can be easily archived, that would result on a cut of power supply! My car is using the shaper to sacrifice charging power to use on the house.

The problem is when my house is consuming so mutch power that the shaper goes under 6 amp. The relay turns of... With the power going up and down the relay is always going on and off... That's not healthy! That's way I suggested that when the power goes from under 6 to over 6, it should wait for some amount of time (determined by user) before turning the relay on again! Reseting the time whenever it goes under 6amp.

KipK, 2023-04-01:
We're talking about Divert mode not shaper here.

But shaper already pause for 5mn when it goes under 6Amps.

soloam, 2023-04-01:
> We're talking about Divert mode not shaper here.
> 
> But shaper already pause for 5mn when it goes under 6Amps.

In what version was this introduced?

KipK, 2023-04-01:
mhh checking at the code now and seems I have forgotten this part yes. Was pretty sure I've implemented it, thanks for pointing it.
I'll add this, and make the pause time configurable.

Perhaps smoothing the shaper input just for getting out of pause could be usefull too.
What would be the best formula for this ? @pdhoogh seems to knows his maths :)

soloam, 2023-04-01:
The most important part is that turning off the relay should be instant, no smoothing factor, because smoothing will only delay the going off, resulting on a over power consumption, and cutting in power supply!

KipK, 2023-04-01:
What I've done for now, ( will PR it in a few ) is pausing 5mn when it reach <6amp,
then the timer is reset even in pause when it receive value < 6 amp. 

It should be safe enough to consider the power is now stable

jeremypoulter, 2023-04-01:
If you want to turn off immediately, set the decay to 1 and the main charge time to 0, then tweak the attack to suit your requirements.

KipK, 2023-04-01:
[#590](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/590) 
temporary fix for now until I redo this differently

pdhoogh, 2023-04-01:
I am having a party now, so brief: agree with @jeremypoulter.

pdhoogh, 2023-04-04:
Currently trying with @jeremypoulter suggestion. Seems to work fine so far. No need for the requested feature, can be achieved with filter settings:
[image]

pdhoogh, 2023-04-07:
@jeremypoulter suggestion using filter works great. 
[image]
You can see how available current jumps up and down and there might have been charge starts that would have stayed on for 10 minutes because of the minimum charge time setting.

With the filters and the minimum charge time set to e.g. 30 seconds, the desired behavior is implemented.
You can see how fast the smoothed value used for switching ON/OFF is following the downramp of the unsmoothed available current. But is slow to pick up a rising. Excellent. 

I just did not understand the filter. Now I do. @KipK with GUIv2 solved that lack of understanding risk perfectly for me. With the old GUI, a bit more doc would be nice.

KipK, 2023-04-07:
We will probably switch to uiV2 for next major release anyway. 
@pdhoogh That would be nice if you could test the new input filter and propose some good default values for the presettings ( the one I've set for now are just empirical based on your explanation but we can probably improve this. 
The presettings I've made only change the filter values, but perhaps they should include the minimum pause time too? ( It default to 5mn for now perhaps it's a bit too much used with the filter.

pdhoogh, 2023-04-07:
Good thinking. Can we do this discussion on your new filter thread?

closed 2023-04-07 by pdhoogh

KipK, 2023-04-07:
Sure that

larroy, 2023-06-16:
the only problem with the filter is that seems to update only when mqtt messages are received...

pdhoogh, 2023-06-16:
I believe this is the whole point of the input filter, in the  new version it is made independent of the sample period. Meaning that the time between samples is now taken into account in the filter calc. When the sample period varies, the calc makes the behavior of the filter independent of that period and fixed in time. Meaning that e.g. a one minute time constant remains a one minute time constant behavior, independent of sample period and calc period. It seems to work pretty well, I've been using it for 2 months now.

larroy, 2023-06-30:
@pdhoogh yes you are correct. In my branch with the new divert logic this works as expected.