OpenEVSE/openevse_esp32_firmware #588: Pilot Limit out by 3 amps

dmelliott20, 2023-04-01:
Just took delivery of a new bmw ix3 that shows the limit of charge that is being requested by the pilot from openevse.

it is consistently 3 below what is set….

set limit 6 amp - car does not charge
set limit 9 amp - car reports 6amp limit
Set limit 20 amp - car reports 17amp limit

I assumed this was just my Audi behaviour previously as I couldn’t see the pilot Limit in the Audi interface.

any ideas?

jeremypoulter, 2023-04-01:
Would you happen to have anything that can / the ability to measure the duty cycle of the pilot signal?

dmelliott20, 2023-04-01:
I think my multimeter has that function.

Leave it with me and I’ll do some tests.

Dave

jeremypoulter, 2023-04-01:
Great, FYI <https://openevse.dozuki.com/Guide/Testing+Basic+and+Advanced/12#s87>

dmelliott20, 2023-04-02:
Yeah so it appears I’m short on duty cycles

16amp 23.9%

20amp 30.6%

32amp 50.5%

This is the official evse euro kit build too.

Dave

jeremypoulter, 2023-04-02:
Thanks, they do indeed seem to be out. I will check on my unit to see if I see the same. What OpenEVSE firmware do you have installed?

dmelliott20, 2023-04-02:
Here’s the details…..

this is the unit I purchased https://shop.openenergymonitor.com/openevse-wifi-ev-charging-station-7kw-european-kit/

[image]

dmelliott20, 2023-04-03:
Thank you, Let me know if I need to test anything further.

glynhudson, 2023-04-03:
Interesting, I'm running controller FW V8.2.1 which is identical to V8.2.2 apart from the front push button not being enabled on all builds, and I'm not able to replicate. I'm running UI V2, but this should not play any part in the pilot signal. 

[image]
 
What EV are you charging? Are you able to test with a different EV?

dmelliott20, 2023-04-04:
Yes since discovering it I’ve tried 4 other EV’s now. I’ve always noticed it charged both my audis at 29.5amps when set to 32 but I just assumed it was the cars. It wasn’t until I plugged in the new bmw that it displayed the limit on the dashboard of 29amp and I noticed.

assuming I’ve measured right too. I just measured from cp to ground?

ruddenchaux, 2023-04-04:
@glynhudson a similar issue was reported on https://community.openenergymonitor.org/t/nissan-leaf-pilot-signal-not-being-followed/22584 where the tested EV are a Nissan Leaf and Hyundai Kona.

dmelliott20, 2023-04-04:
Yes that’s the exact same issue I’m having.

glynhudson, 2023-04-04:
Thanks for testing, that's very useful. I will investigate further

dmelliott20, 2023-04-04:
Thanks, to confirm tested on the following

Audi Etron GT
BMW I3
BMW ix3
Audi Etron sportback
Jaguar Ipace

all show the same result within 0.5amps

dmelliott20, 2023-04-04:
@glynhudson are you testing on the same hardware as I have detailed above?

dmelliott20, 2023-04-04:
Yes same behaviour...

[image]

dmelliott20, 2023-04-30:
Closing this issue as it was down to faulty hardware.

swapping out the openevse board fixed the problem.

thanks everyone

closed 2023-04-30 by dmelliott20