OpenEVSE/openevse_esp32_firmware #591: OpenEVSE nuisance trips, only when PV production is high

Jeedomguy, 2023-04-01:
Hello,

I bought an Open EVSE Advanced Series Kit one year ago, and it works pretty well so far, except for an issue that I can't solve:
I'm trying to charge as much as possible when my rooftop PV is producing, but I experience charger tripping when the PV production is very high (above 2500W for an installation of 4500Wp).
The charger never trips at other moment. 
Everything goes well when charging at night.
Everything goes well during the day when the PV production is low.
Nuisance trips happens only when the PV production is high.
I tried to open the circuit breaker of the PV system to see if it solves the issues immediately after, but the problem remains.

I looked at the logs, but nothing. 
The logs show that the status switch from "Charging" for 30 to 60 seconds, then "EV Connected" for 10 to 20 sec, then "Charging" again for 30 to 60 sec, and so on 4 or 5 times until it stops trying and stays in "EV Connected".
There is no message issues appearing on the car (Renault Zoe from 2017).

Do you think there could be voltage issues or earth issues on my network when PV is at full power, but without appearing on OPENEVSE logs?

See the logs attached showing the sequence.

Thanks in advance for your help.

[image]

glynhudson, 2023-04-02:
Can you check to see if there is an error message displayed on the LCD when the unit trips?

Jeedomguy, 2023-04-07:
No there is no error message displayed on the LCD.
Note that I hear a slight noise/vibration in the charger when it's in charging mode. Not sure if that can be linked or not.

Jeedomguy, 2023-04-07:
> do you have a graph of your published pv data used by divert mode ?
> How is the frequency of the data publishing ? Are they coming in a regular interval ?
>
> New input filter will handle non fixed sample rate, but current implemented one needs fresh data each 10sec.
>
> May I see your divert parameters ?

Hello,

I'm not using Divert mode.
I'm using my Home Assistant (an open source one called "Jeedom") which monitors the PV production every minute, and send a new charging power command to OpenEVSE accordingly.
See attached the chart and logs for the last 30 minutes where I had again 3 short disconnections.

Thanks,

Damien

Jeedomguy, 2023-04-07:
[](url)
[image]

KipK, 2023-04-07:
Ok I see you use 4.1.7 it has a lot of wifi disconnection problem because a
bug with Mqtt. Can you try with latest dev build ( v2_gui ) and post a
screenshot the monitor/command tab when it trips out. And can you use the
export data from the Développer tab and post them too ?

I use Jeedom too. You can reach me on Jeedom community if needed.

Jeedomguy, 2023-04-07:
Thanks Guillaume.
Let me that this weekend and until I get the issue again and I’ll update you with the logs.
Merci !

closed 2023-07-05 by glynhudson