OpenEVSE/openevse_esp32_firmware #592: Power sharing

kallisti5, 2023-04-02:
Fun idea.  As more and more people begin to have more than one EV, OpenEVSE should look into powersharing.

Example workflow:
  * Set a "main" OpenEVSE unit
  * Other OpenEVSE units can "join" it creating a "pool" of OpenEVSE units
  * A maximum service level can be defined for the pool (32A for example)
  * OpenEVSE units take turns charging, or divide up the service (2 units charging, offer 15A each, etc)

For maximum flexibility, units should communicate to each other.  While in pooled mode, loss of network connectivity means members refuse to charge.

There are safety concerns, so it would be best to set big overheads (for example, when sharing a circuit only use 80% of it, etc)

KipK, 2023-04-04:
I don't think that it's evse job to handle a fleet, but more the task of an external service 
Wouldn't it be more appropriate to build a simple external load balancer script or using Hass/Jeedom/whatever ? There's an api for that on evse to use with.

KipK, 2023-04-04:
Even just tricking the main live power mqtt feed used by Shaper would work nice I think

fhteagle, 2023-04-05:
Doesn't OpenEVSE's OCPP implementation already handle this elegantly? Not sure as I have never used it myself, but...

https://www.chargelab.co/industry-advocacy/ocpp

kallisti5, 2023-04-06:
Overall, OCPP does some of this as of OCPP 1.6+.  However, all OCPP implementations seem to require a central server, database backend, account management, etc.

This all seems like overkill for just wanting two chargers to share a single source and balance.

fhteagle, 2023-04-08:
This can definitely be done in HomeAssistant, as well. Let me know if you want to to go that route and I can probably get you through a recipe to do so.

KipK, 2023-04-08:
Node-red could fit if you don't want to script something in Python or node.js

chris1howell, 2023-04-08:
OpenEVSE has been capable of power-sharing for a very long time, it just takes a fair amount of technical knowledge to pull off.

Many of the big guys are offering simple power-sharing. Tesla offers up to 4 HPWCs connected vis RS485. Clipper Creak Share2 (unspecified 2 wire digital serial). Enel X offers Load Share Group functionality via Juice Net etc. 

I think a some point, OpenEVSE needs to offer similar functionality without a complex external service/script. 

IP Multicast would be a good way to exchange information, such as group name/ID, current pool size, number of stations, priority, heartbeat, station status, power consumed, loss comm safe rate, etc.

KipK, 2023-04-10:
ArduinoMongoose has no udp server nor multicast ready yet I think. 

But multicast won't be a solution here, there's cases where all OpenEvse won't receive the same commands, and it will be tricky to manage connection states over multicast.

-> we only have 12A available ( defined by Shaper or Divert ) for 3 openevse. We need then to only activate 2 of them at 6A and disable the third one instead of disabling everybody. ( thinking of that, how do we define priority rules ? ) 
-> perhaps some wants to charge full speed one EV first and then switch to the other one when done. 

It needs coordinating each of them separately, http api could be use for that but a real sock connection should be better to handle connection states. 

There's some setup to add:
  - clients ip / pass on server side, or server ip / pass on client side, not sure wich one is best yet.
  - priorities depending of current available 
  - max vehicles we can charge at the same time
  
some question to solve:
 - what happend when network is down or some clients are unreachable ?
 - How do we know they are really powered off or just unreachable but can still draw amps on their own ? 
 
I think for that a default rule of disabling the evse if not connected to the server or to the network would solve booth question
Shaper already handle this if not receiveing data for xx seconds but it's not instantaneous

cjbarth, 2024-12-27:
> * what happend when network is down or some clients are unreachable ?
> 
> * How do we know they are really powered off or just unreachable but can still draw amps on their own ?

You don't have to reach everyone, you only have to be able to reach the meter that tells you current consumption and you have to know max consumption, and then you have to adjust accordingly.  I've posted some code [here](https://github.com/OpenEVSE/openevse_esp32_firmware/issues/940#issuecomment-2543119167) which I'm thinking about update with just this scenario to show what would happen.

chris1howell, 2024-12-27:
So the suggestion is do nothing except use a proprietary API meter reading
instead of a standard MQTT meter reading and all the stations will just
figure it out on their own?

cjbarth, 2026-08-27:
It can be a standard MQTT meter reading, but something has to host the MQTT and everything else has to reach it. In a no-internet scenario, that becomes problematic. If the meter _also_ has its own MQTT server, that makes it a little easier as this can all be done over a local network. I suppose if the EVSE could run MQTT as well, that would work.

The idea I was presenting is not predicated on each EVSE being able to read some custom data, though I don't see why that would be hard to do; there aren't many meters, and their payload is easy to figure out. The idea is that all this should work without someone dispatching commands. It should be a "pull" configuration, not a "push" configuration. In a "push" configuration, if the pusher goes off-line or is unreachable, everything either stops or over-draws. In a "pull" configuration, each device will pull as much as it can verify that it can. If it can't verify anything (meter is unavailable), it draws nothing.

One shouldn't have to maintain a meter + EVSEs + some other computer, to get this to work. Eliminating "some other computer" and "internet" while still having power sharing and load balancing is the goal.

closed 2026-09-03 by chris1howell