OpenEVSE/openevse_esp32_firmware #594: Incorrect Elapsed and Delivered when EVSE status is set to enabled before plugging the EV

ruddenchaux, 2023-04-08:
When I set the status of EVSE to enabled and sometime later I plug the EV, I see both Elapsed and Delivered are wrong computed. It happens on V2 GUI, but the issue also lives on V1 GUI, I think.

Background
- No EV plugged
- The EVSE status is disabled/sleeping

Steps to reproduce
- Set the EVSE status to enabled
- Wait sometime (more time, more noticeable the issue is, I think)
- Plug the EV

Current Behaviour
Both Elapsed and Delivered are wrong (In my opinion the counter start when the user set the EVSE status to enabled).

jeremypoulter, 2023-04-08:
The charging session starts from when the charging session starts and finishes when the vehicle is disconnected. I think this makes the most sense as you can connect/disconnect multiple times without enabling/disabling the charger, and likewise the charger can be enabled/disabled multiple times during a charging session (eg when using divert)

ruddenchaux, 2023-04-08:
Thanks, @jeremypoulter. I know what you are saying but I wouldn't want the charging session starts when I enable the charger and the EV is not connected, otherwise, when I connect the EV, "Elapsed" and "Delivered" will be greater (because the charging session is already started).

jeremypoulter, 2023-04-08:
I am not quite sure I get your problem, can you give some concrete examples/screenshots and what you would expect the values to be?

ruddenchaux, 2023-04-08:
Sure,

1. The charger is disabled and the EV is not connected (shaper enabled, eco and divert disabled, no limits and no schedulers) 
2. I enable the charger
3. I wait sometimes (eg half an hour)
4. When I connect the EV then I see "Delivered" doesn't start by 0 (eg 1kwh) and "Elapsed" doesn't start by 00:00:00 (eg 19:50:23)

Why?

PR #596 (2023-04-12): loop energyMeter.update() even when not charging
 ( was not rotating data correctly )

probably fix #594
closed 2023-04-12 by jeremypoulter