OpenEVSE/openevse_esp32_firmware #595: Add some new /config flags & /status property

KipK, 2023-04-10:
/config:
- wizard_passed 
  feature: set default to 0, when wizard has been passed once, set the flag to 1.
 comment: find a way to not display the wizard for upgraded setup. ssid is not sufficient as it doesn't solve ethernet module. Can we detect the whole config file is still default ? 

- default_state: 0 | 1 ( active | disabled )
 feature: allows to choose default evse state (active or disabled) when no override/claim has been setup.
comment: I think EVSE fw should also be set to disabled by default, not sure this is the case or if it has any collaterals.

- divert_type: -1 | 0 | 1 ( unset | solar | grid i/e )
 feature: store divert type. Remove needs of mqtt topic ( needed for http push ) ( -1 needed to import previous fw behavior ) 

- vehicle_data: -1 | 0 | 1 | 2 ( unset | http push | mqtt | tesla )
 feature: store vehicle data source. remove  needs on checking tesla credentials or mqtt feed. 

/status:

- mode_auto: 0|1 
  feature: match the auto mode from UI, if there's no "state" property set in /override , then auto is 1
  goal: remove some boilerplate code from UI side & simplify other integrations to match current UI2 use case. 
  comment: probably also need an endpoint on api to activate auto ( if called, it remove /override "state" property if there's other properties in the /override claim, or clear the override if there's only the "state" prop set. For now it's done on UI side. )

jeremypoulter, 2023-05-02:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/src/evse_man.cpp#L169 I think

KipK, 2023-05-02:
I'm not sure about the mode_auto thing for now. 
Current UX was made with the constraint of beeing compatible with old UI.   
I'm removing it from this issue and postpone it after UI migration, there's probably things we can rethink a bit.

KipK, 2023-05-02:
#618 achieve all those & seems to work well here.

PR #618 (2023-05-21): New settings / flags (#595)
FIxes #595
needs UI branch "newflags"

Not to be merged before UI migration

done: 
 * wizard_passed ( fw + ui )
 * divert_type ( fw + ui ) 
 * vehicle_data_src ( fw + ui )
 * default_state (fw + ui )
 * set AP ssid & password ( fw + ui ) 
 * added /restart {"device": "gateway|evse"} to http api & mqtt + restartEvse() method in evseManager ( close #479 ,
closed 2023-05-21 by jeremypoulter