OpenEVSE/openevse_esp32_firmware #599: Total usage doubled after upgrade?

TommySharpNZ, 2023-04-14:
I just upgraded to the new version to get the cool new UI and also the additional usage counters...
I notice that Total Usage doubled after the upgrade. Is this intentional or some issue?

[image]

KipK, 2023-04-14:
Strange, it's supposed to import once the total value from OpenEvse module .

You can reset the counter and  choose if you want to import old values or not :  https://openevse.stoplight.io/docs/openevse-wifi-v4/718f0f0931c65-reset-energy-meter

danpadure, 2023-04-26:
how did you get that graph?

TommySharpNZ, 2023-04-26:
> how did you get that graph?

That graph comes out of Home Assistant....

KipK, 2023-04-26:
@TommySharpNZ is the total only wrong on home assistant or also on openEvse ?

TommySharpNZ, 2023-04-26:
Oh yeah, definitely wrong at the source in OpenEVSE. Seems to have happened when upgrading... But I'm not too fussed as I really wanted the Daily counter so am very happy with that :-)

[image]

KipK, 2023-04-26:
seems it imported twice, I wonder how it could. 
Can you try to reset the counter with import: true to reimport again as linked [above ](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/599#issuecomment-1507938622)?

closed 2026-09-21 by github-actions[bot]