OpenEVSE/openevse_esp32_firmware #600: Can not update from 4.19 to 4.2.2 with v2 gui

ausil, 2023-04-17:
I have two OpenEVSE's one is older and has a huzzah32 module in it and I have upgraded it to 4.2.2 just fine, I have a newer unit that has a openevse_wifi_v1 module and when I click on upgrade nothing happens.  the changer sees the new firmware it just will not load up the window to upgrade

[image]

KipK, 2023-04-17:
That's a little bug on UI I have fixed since
You can upgrade using Curl command:

curl -F firmware=openevse_wifi_v1.bin http://openevsze.local/update --progress-bar | cat

ausil, 2023-04-17:
```
$ curl -F firmware=openevse_wifi_v1.bin http://openevse2.ausil.us/update --progress-bar | cat
####################################################################################################################################################################################################################################### 100.0%curl: (56) Recv failure: Connection reset by peer
```
and still shows v4.1.9

closed 2023-04-17 by ausil