OpenEVSE/openevse_esp32_firmware #601: Bricked WiFi Modules after 4.2.2 firmware update?

ve6rah, 2023-04-17:
I have an OpenEVSE on version 7.1.3 with the ESP32 wifi module.

I downloaded [openevse_esp32-gateway-f_gui-v2.bin](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/v4.2.2/openevse_esp32-gateway-f_gui-v2.bin)
and updated it through the GUI.
After the update, the WiFi module never came back online. The display showed a message saying that the firmware update was complete, and just sat there for over 20 minutes. I finally power cycled the OpenEVSE, and the lights on the wifi module never returned, and the EVSE acts as if no wifi module is installed. The wifi module does not connect to my network, nor does it create its own hotspot.

Thinking it must be something with that one wifi module, and as I happen to have a spare on hand, I downloaded [openevse_esp32-gateway-f_gui-v1.bin](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/v4.2.2/openevse_esp32-gateway-f_gui-v1.bin) and updated that through the gui of the second wifi module.
This time it hung at 98% on the LCD display, and the wifi module dropped from my network. After 20 minutes sitting at 98% I power cycled OpenEVSE and it acts exactly as in my first case, no lights, no wifi, no indication on the display that a wifi module is even installed.

ve6rah, 2023-04-17:
Well, looks to me like this is my fault. I used the wrong binary (In my defence, the docs could be a LOT clearer than "use the right file" and actually make it easy to see which one is right!)

So I'm going to try manually flashing with esptool to get back on track.

KipK, 2023-04-17:
You should be able to get it back by flashing with esptool yes. 
May I know what board you have to understand how you took the wrong fw ?

ve6rah, 2023-04-17:
I have the normal wifi module that comes with all the OpenEVSE kits at this point. It has an ESP32 chip on it, so I chose ESP32.

There's a pretty good description of which one you need down in the 4.1.5 release notes, but the latest version just says to make sure you use the right one, and doesn't tell you how to know which is right.

I'm not currently having much luck with esptool. Keeps timing out waiting for packet header. still trying.

KipK, 2023-04-17:
You have to keep pressed he reset button ( right one nearest to the USB port ) , then press left button, the release right button. 
It will stay in flash mode

KipK, 2023-04-17:
Once everybody have switched to UI V2 it will be easier, as firmware update is now automatic from Github files and doesn't needs users to download manually the correct file. 

This will prevent those kind of mistakes.

ve6rah, 2023-04-17:
There's no USB port on the wifi module. There is a button marked reset, and another marked GPIO 0
I tried as you said and held reset, tapped GPIO 0 then released reset. This didn't seem to have any effect.
I also tried the reverse, holding GPIO0, tapping reset, then releasing GPIO0, that didn't work either.

While messing around with the buttons, a couple of times I've had the green LED turn on (and stay on) but I'm not sure if that signifies anything at this point.

ve6rah, 2023-04-17:
Aha! I finally got it.

Hold GPIO0 while plugging in the usb programmer, then release. Then Flash.