OpenEVSE/openevse_esp32_firmware #603: Websocket disconnected

FatJuan, 2023-04-21:
I upgraded to openevse_huzzah32_gui-v2 and find the strange behaviour of Websocket disconnected, waiting for reconnection this is difficult for me because I often want to charge my car when I am not in the house using excess solar.  When it is charging via solar the actual message "Websocket disconnected, waiting for reconnection" has no effect on the car continuing to charge, perhaps because all of that is on the local network? The time freezes and obviously the time cannot be updated with no connection? I tried to load the openevse_huzzah32_gui-v1 but was not successful . Thanks for any comment 
[image]
[image]

KipK, 2023-04-21:
You need to forward websocket headers when proxying to your internal network

KipK, 2023-04-21:
However I strongly recommend you to not gives direct access to your evse over the internet. This is a microcontroller with memory constraints, it's easy to find security flaws on those kind of device. Use a VPN instead to connect to your home services, it's far more secure and should gives you easier setup.

FatJuan, 2023-04-22:
I am not at present giving access to the evse over internet. I am looking locally. The only reason I mention it is because this is different behaviour to the previous version I was using 4.1.7 on openevse_huzzah32. Refresh page and it is all there. Thank you for your concern.

KipK, 2023-04-22:
You mean it always displays the websocket disconnected message even on local network ? 
That's clearly not normal. 
You browser needs to connect to Wifi module over websocket to get status data update. 
On previous Web UI, it was polling /status each second, so it could work even if weboscket was not working.

Now we need to find why websocket connection fails on your side.
Can you check the browser console output ( developper tab )  if there's error msg ? 

What browser + OS are you using ? 
Have you tried on your smartphone too if it's the same ?

FatJuan, 2023-04-22:
I have found the same behaviour on Firefox and Safari Mac (macOS 13.3.1) One thing that may be of note I am signing in using http if I try https I get cannot connect to the server on both safari and Firefox same on iPad. Any help? Add I have said if I refresh page all ok
Thanks

KipK, 2023-04-22:
There's no https on openEvse wifi, only http://

I don't understand, you mean if you refresh everything is ok? It doesn't display the websocket msg? 
Then it's probably just when your browser come back from sleep but it should reconnect after few seconds automatically and remove the msg.

FatJuan, 2023-04-23:
It only seems a problem on Safari. It's ok with Firefox. Thanks for your help.

KipK, 2023-04-23:
I keep it open, I need to investigate this. But Safari has been tested ok already. 
Don't you have a kind of VPN on your browser? I know embedded VPN from Vivaldi browser as example gives some issues with websockets.

FatJuan, 2023-04-23:
I will do that but I am not very capable.
I have a choice of
Open Page With
User Agent
MacBook Air
Service Workers
Web Extension Background Content
Experimental Features
Enter Responsive Desigh Mode
Show Snippet Editor
Show Web Inspector
Show JavaScript Console
Show Page Source
Show Page Resources
Start Timeline Recording
Start Element Selection
Empty Caches
Enable Intelligent Tracking Prevention Debug Mode
Disable Images
Disable Styles
Disable JavaScript
Disable Extensions
Disable Site-specific Hacks
Disable Local File Restrictions
Disable Cross-Origin Restrictions
WebRTC
Allow JavaScript from Smart Search Field

What would be useful? Thanks

FatJuan, 2023-04-23:
Hi I have moved to another room served by different AP it seems perfect no issues.  I think it is something to do with my AP configuration. Sorry about that!

KipK, 2023-04-23:
It's the JavaScript Console , copy paste the output .

FatJuan, 2023-04-23:
[Log] opening socket (index-701bb6da.js, line 1)

[Log] connected to websocket (index-701bb6da.js, line 1)

[Error] WebSocket connection to 'ws://192.168.1.3/ws' failed: The operation couldn’t be completed. Connection reset by peer

[Error] Socket encountered error:  – undefined – "Closing socket"

	(anonymous function) (index-701bb6da.js:1:3064)
[Log] Socket is closed. Reconnect attempt in 1 second. – "" (index-701bb6da.js, line 1)

[Log] opening socket (index-701bb6da.js, line 1)

[Log] connected to websocket (index-701bb6da.js, line 1)

[Log] No msg over websocket for 211 sec, restart websocket (index-701bb6da.js, line 1)

Hope this is of use

KipK, 2023-04-23:
Seems like  a firewall rule is blocking/dropping  websockets when you try to reach 192.168.1.3 ( is it openEvse IP or a forwarded port ? )

FatJuan, 2023-04-23:
It is the OpenEVSE IP. I am using Safari in private window mode. I will see if I can find anything in the router that may impede.

FatJuan, 2023-04-23:
Looking at my Router Draytek 2927 the config for the firewall is default I think
Data Filter | Enable Start. Filter Set. Set #1
Allow pass inbound fragmented large packets (required for certain games and streaming) Enabled
Enable Strict Security Firewall Enabled
Block routing connections initiated from WAN Disabled IPv4 Enabled IPv6
Any help?

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none;">

closed 2023-04-25 by FatJuan