OpenEVSE/openevse_esp32_firmware #604: RAPI codes $FE and $FD no longer work in 4.2.2

omriasta, 2023-04-23:
I usually use these codes through mqtt but I have tried sending them in the user interface in the developer panel as well. The codes register, I get an OK response in the web interface. It appears to change the status or do something but then immediately goes back.
I have tried this under both the old and new web ui. Downgrading to 4.1.7 fixes this issue. Though staying at 4.1.7 and upgrading just the new web ui also breaks this.

jeremypoulter, 2023-04-23:
This is expected behaviour. The latest version is now more forceful in making sure the EVSE is in the state that WiFi firmware expects. With the v4.2.2 release the HTTP API should implement all the functions of RAPI (and more) so we made the decision to make that change. Also note using RAPI has been deprecated since the v4 release.

closed 2023-04-23 by jeremypoulter

omriasta, 2023-04-23:
Understood, is there an option to enable/disable using mqtt?

KipK, 2023-04-23:
use the claims or override endpoint on mqtt:

https://openevse.stoplight.io/docs/openevse-wifi-v4/c03364fd24abd-mqtt

omriasta, 2023-04-23:
Got it, thank you!!