OpenEVSE/openevse_esp32_firmware #606: Extremely frustrating update process

j256, 2023-04-28:
I've got an OpenEVSE but I've got questions about updating.

Found "Firmware update" in the manual.  It mentions the bin files and I see the 4.2.2 which looks to be the stable release.  But I see v1 and v2 bin files but no description of what those are or how I determine which to use.  I also can't see anything about v1 or v2 on the web interface itself.  I'm worried that I will brick my unit if I load v2 on a v1 or something.  I don't see a picture of a v2 for comparison?

Some help?  Thanks,

KipK, 2023-04-28:
It's written at the top of the release description:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/v4.2.2

closed 2023-04-28 by KipK

j256, 2023-04-29:
Thanks for the response @KipK .  I see the mention but I still have questions.
* How do I determine which one I have now?
* What are the differences?
* Is v2 some sort of UI upgrade?

KipK, 2023-04-29:
V2 is new user interface. If you have it you would see it then, it's completely different