OpenEVSE/openevse_esp32_firmware #608: To Do for 5.0.0

KipK, 2023-04-28:
Opening a thread here on short term roadmap for next major release 5.0.0

1- Switch to GUI V2
 * Most important point, we get more & more users feedback confused by the v1 v2 builds + maintaining booth for added features is a pain. 
 * Do we still have some missing points / buggy features in the main interface ? 
 * build naming convention:
   . remove _gui-v2 at the end of the filename & remove v2_gui release tag
   . this will probably break update from github on UI as it is implemented. Assume it as a last time issue, or find a way to prevent it.
   . we could take the opportunity to use our own generated json file for releases/builds 

2- new config properties ( #595 )
 * there's a bit of work here to prevent braking previous setup, i have some work in progress on this I need to finish
 * remove all the MQTT field dependecies as we have HTTP push as a data source alternative
 
3- new InputFilter ( #590 )
 * seems to works ok for those who have tested it. Needs to be reviewed a bit more. 
 * simpler to understand how to setup than previous filter ratios values and has pre-settings in the GUI
 * todo: refine presettings values

4- Issues? ( todo: list here issues ID that should be solved for 5.0.0 )
* #597

jeremypoulter, 2023-05-01:
Moved all this to a project: https://github.com/orgs/OpenEVSE/projects/2/views/1

closed 2023-05-01 by jeremypoulter

KipK, 2023-05-02:
Page not found, is it private ?

jeremypoulter, 2023-05-02:
Ah, yes it is and I can not change it. I have invited @KipK , but @chris1howell could you change all the projects to be public?

chris1howell, 2023-05-02:
All projects are public. I also added Discussions for the repository.