OpenEVSE/openevse_esp32_firmware #621: Restart required after Wi-Fi connection to update NTP

glynhudson, 2023-05-03:
After connecting a new WiFi module to a WiFi network the module requires a restart before NTP will set the time. 

How long should it take for NTP to update? Possibly I didn't wait long enough, but I was waiting about 20-30s with no NTP update before restarting

KipK, 2023-05-04:
That's odd it's supposed to be instantaneous when you select NTP from the drop down selector.
When you switch from Manual to NTP doesn't it set correctly ? 

Does your EVSE got working DNS from dhcp over your wired network ?

KipK, 2023-05-04:
#624 add a time event publish after ntp update. It was missing, it probably is the reason ( but time should have updated after 30 sec max in your previous test )

KipK, 2023-05-04:
I think I should also move time setting after the wifi redirection step in the wizard but that doesn't explain your case with ethernet.

glynhudson, 2023-05-04:
>That's odd it's supposed to be instantaneous when you select NTP from the drop down selector.

I've not touched the NTP selection since NTP is selected as default. 

>When you switch from Manual to NTP doesn't it set correctly ?

I've not tested this 

>Does your EVSE got working DNS from dhcp over your wired network ?

Yes, DHCP and wired connection works fine. I also experienced the same issue with WiFi module

KipK, 2023-05-04:
can you merge #624 and test the generated daily gui-v2 build if it fixes it?

closed 2023-05-04 by glynhudson

KipK, 2023-05-06:
@glynhudson is it fixed ?

PR #624 (2023-05-04): fix time not updated immediately after NTP change
time was lagging behind after ntp update. 
perhaps fix #621