OpenEVSE/openevse_esp32_firmware #628: Wifi losses connection - does not recover.

irekzielinski, 2023-05-06:
Hi everyone.
I have the latest version of OpenEvse (2 months old).
Problem is occasionally unit will lose wifi connection (not accessible via web interface, MQTT dropped).
I have two access points - one simple 2.4Ghz one dual 2.4 + 5Ghz, issue is seen on both. Both very close, -55dBm typical.
Today I wake up with car not charged:
- I set energy limit to add 9kwh overnight
- I had also a speed override set to 12 amps (out of max 32)
- my Nissan Leaf has a timer to come on economy 7 time (1:30 am)
- unit lost connection to wifi at 00:03 and car failed to charge
- unit LCD appeared like all is good (cycling trough various displays as normal)
- after reboot of the OpenEvse wifi reconnected - but system max amps was reset to zero

Firmwares:
OpenEVSE | 8.2.2.EU 
OpenEVSE Wifi | v4.2.2

Any advice how to resolve it (I know that ethernet gateway is an option),
As wifi unit appeared to do something (driving LCD display for example) - maybe there could be an option to auto-restart esp32 when wifi or mqtt is lost for more than 5 minutes or so?

KipK, 2023-05-06:
if system max_amps is going to zero, it seems there's a problem on evse module side.
Don't you have any errors displayed in monitoring/safety ?

irekzielinski, 2023-05-06:
Lost wifi multiple times - but this drop to zero after reboot was fist time ever.
No error displayed in "monitoring/safety" - every counter is zero.
I have shaper enabled via MQTT (if related) but load is not even close to maximum.

KipK, 2023-05-06:
Next time it happens with the max current at zero go to Config/Developer and export all data and post it here please

KipK, 2023-05-06:
Also I'll compile you a debug enabled version later in the week end, try to check in the dev console if there's something suspect

KipK, 2023-05-06:
@irekzielinski try this : https://drive.google.com/file/d/1HcXIg0_0oFXBT_aKCMD5Gi8JqvAZsbUF/view?usp=sharing
i've enabeld debug for evse_manager and mqtt  but monitor to not spam the console, let's see if it's sufficient.

irekzielinski, 2023-05-06:
I am currently on v2 firmware - your file has v1 in the name: is this safe to flash/downgrade?

KipK, 2023-05-06:
it's is gui v2 yes, wifi module is still open_evse_wifi_v1

irekzielinski, 2023-05-06:
Thank you - flashed

[image]

So the next time unit will disconnect from wifi - what should I do?
Reboot and capture content of Developer/Console/Debug content?

KipK, 2023-05-06:
yes and /developper > Export all Data

irekzielinski, 2023-05-06:
problem is that when mqtt dies, I am not able to connect to web interface as well
the only sign of life is that LCD is displaying "usual" information, cycling trough various options.
I already tried two different access points (one on channel 1, second one on channel 6). I live in detached house - no significant interference. Freeze also happened at night - when no-one was using wifi.

What about:
- when device detects loss of mqtt/wifi it will save console to flash memory (so it can survive a reboot)
- would it be possible to add an option for auto-reboot if mqtt/wifi is down for 5 minutes or so? this might be very helpful as waking up with car not charged is a bummer

irekzielinski, 2023-05-06:
Did some more digging. I have some other esphome devices connected to the same WiFi.
And noticed that wifi was unstable overnight for around 2 hours:

This is one of the sensors in home assistant (ESP32-wroom dev kit running esphome firmware)
[image]

The difference is that esphome device was disconnecting briefly and since then was solid, but openevse lost wifi until reboot.
So maybe to try to reproduce the issue - one could mess with wifi signal.

closed 2026-09-21 by github-actions[bot]