OpenEVSE/openevse_esp32_firmware #630: How to control/set divert mode in OpenEVSE using http api?

rbhollabaugh, 2023-05-06:
I was sent here to ask this question from the OpenEVSE forum.
I am trying to use the http api (no mqtt) to send grid_ie and solar production numbers to the unit so it can modulate charge based on either or both of those numbers.
I am using this as a guide. Is this correct information?
https://openevse.stoplight.io/docs/openevse-wifi-v4/21c8a9ac8dd46-update-external-data-to-evse-status

The http command is successful in getting these 2 numbers to the unit but it does not modulate the charge. The max charge is set to 40amps. When divert mode is on the charge starts out at 40A and then slowly goes down to 6A over several minutes.
OpenEVSE firmware is 8.2.2
OpenEVSE wifi firmware is v4.2.2 on an esp32-gateway board.

rbhollabaugh, 2023-05-06:
An update...
OK in playing around with the self production (production and export) settings I see a problem. Even though mqtt and Energy Monitoring (emoncms) are both disabled, it insists I enter a topic. There is a pre-loaded emoncms topic there that I never entered. If I erase it, it gives me an error message and won't let me enable Self production. So it seems the code is insisting I use a mqtt or emoncms server.
And I don't want to use either of these. Just http as in the api documentation. Is the documentation up to date? Do I have to use mqtt broker?

KipK, 2023-05-06:
Yes itnisnthe case for now. I have a PR ( new_flags ) that solve this. It will be merged for next release. 
For now you can enter a fake Mqtt topic to dumb it

rbhollabaugh, 2023-05-06:
Here's what's going on:
Eco mode: enabled
Auto(robot guy): enabled
SelfProduction->ProductionMode: My "solar" production number appears here.
SelfProduction->ExportMode: My "grid_ie" number does not appear here.

In either mode charge does not happen. If I switch to manual mode charge starts at 40A. If I then switch to auto, charge tapers down to 6A after a few minutes.

What are the steps needed to get auto mode to work using http api?

KipK, 2023-05-06:
Waits for the merge of "new_flags" branch it will solve this. 
HTTP push was introduced untested , it works for shaper or vehicle data but divert was testing the mqtt fields to define the divert type so it's perhaps not usable until the divert_type flag is merged soon.

KipK, 2023-05-06:
I've missed this one, you can't use booth solar and grid_ie. Only one of them are they are booth different divert type.

rbhollabaugh, 2023-05-07:
OK - with your help I think it may be working. I won't know until later today after I run the battery down;)
Sending only grid_ie - now the grid power shows up on the SelfProduction->Export mode. Sending both solar and grid_ie at the same time apparently does not allow the grid_ie number to show up.
And putting the dummy feeds into the mqtt topic also worked.
When a cloud goes over I see the charge rate change. Divert is enabled and it is 'active'.
Will let you know the results of this test....

closed 2023-05-14 by KipK