OpenEVSE/openevse_esp32_firmware #638: Olimex ESP32 Gateway. and ESP32-POE

Tinbum1, 2023-05-10:
1)
I've noticed the LED is a little inconsistent in its operation on the ESP32 Gateway.

If booted without wired Ethernet the led will flash and then stay on solid.
If wired is connected then it will continue to stay on.
If wired is then removed it will continue to stay on.
If wired is reconnected it will continue to stay on.
If wired is removed again it will go off, start to flash and then stay off, even though the WiFi is connected.

i.e every other disconnection of the wired Ethernet the led be in the opposite state.

2)
I've also flashed a ESP32-POE and ESP32-POE-ISO with the Gateway firmware (V2 GUI), as the boards look very similar,  but find that it won't switch over to the wired Ethernet when it's connected.  (A wired IP doesn't show up in the network router).

jeremypoulter, 2023-05-10:
1) I am assuming that this is with valid WiFi credentials setup?

2) I unfortunately don't have either of those boards to check, have you checked the network chip connections to see if the same lines are in use? You may need to create a separate build for those boards, also best to raise this in a separate ticket.

Tinbum1, 2023-05-10:
1) Yes, valid WiFi set up, can see it fine in a web browser.

2) Yes, I've looked at the schematics for the Gateway and the POE and they do look the same. Will do :)

closed 2023-06-09 by glynhudson