OpenEVSE/openevse_esp32_firmware #641: Solar Divert: excessive change for pilot value which is not based on smooth current

mathieucarbou, 2023-05-11:
As we see here:

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/8170b699609f2ee4ac3638bbaa30ed7ec5c48f69/src/divert.cpp#L212

The charge rate, which is sent as the value for the new pilot in the claim, is based on the available current and not the _smoothed_available_current.

The problem is that on a shady day with sun and clouds, when the smoothed_available_current is higher because a cloud is moving in, the pilot value is going crazy and is being updated really frequently, updating the charging power for the vehicle.

Is it an expected behaviour or a bug ?

it seems to me that the diver module computes a smoothing available current for a good reason and should use it as the new charge rate. When a cloud moves in, the smoothed_available_current value decreases slowly, which is a nice behaviour.

Currently the vehicle charge power is frequently update when the solar production quickly changes... I guess this is not good both for the vehicle and BMS ?

mathieucarbou, 2023-05-11:
@KipK : fyi we saw that with a friend... same behaviour - excessive updates of the pilot value.

mathieucarbou, 2023-05-11:
If this is a bug, I've raised a PR for a fix: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/642

jeremypoulter, 2023-05-11:
Not a bug, the smoothed value is used for enabling/disabling the charge as switching the relay too often can cause damage to the relay. We use the unsmoothed value for setting the pilot to quickly react to the changing conditions and increase the amount of solar used and decrease the use of grid power as there is little impact of changing the pilot when charging is enabled

mathieucarbou, 2023-05-11:
ok understood. 
So quick changes in power sent to vehicle during a charge won't have any effect / damage on the life of the cells due to a fluctuation of the charge power ?

closed 2023-05-11 by mathieucarbou

jeremypoulter, 2023-05-11:
Shouldn't do, but if anyone has any evidence to suggest it does then we can look at changing