OpenEVSE/openevse_esp32_firmware #643: Remove RAPI

jeremypoulter, 2023-05-14:
Since the switch to v4 and the move to have more of the smart features on the ESP32 RAPI it has always been the intention to remove or limit the user's ability to use RAPI: <https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/docs/rapi.md>

We now are in a place where the HTTP/MQTT APIs offers all the functions that RAPI can provide so I think the time has come to remove the RAPI commands from MQTT and limit the HTTP to RAPI gateway to just the `$Gx` for debugging.

mathieucarbou, 2023-05-14:
I have a Zoe ZE50 E-Tech R110, and this vehicle is subject to the sleep issue.

Ref: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/223

So I added an automation in HA which is sending through mqtt a sequence of `$FD $FE`:

```yaml
automation:
  # 
  - alias: "OpenEVSE: Zoé Bugfix"
    trigger:
      - platform: state
        entity_id:
          - binary_sensor.openevse_vehicle_connected
        from: "off"
        to: "on"
    condition: []
    action:
      - service: mqtt.publish
        data:
          topic: "openevse/rapi/in/$FD"
      - delay:
          hours: 0
          minutes: 0
          seconds: 2
          milliseconds: 0
      - service: mqtt.publish
        data:
          topic: "openevse/rapi/in/$FE"
```

So far, and like reported in the issue in reference, this is pretty reliable.

I've seen this option in OpenEVSE:

[image]

But I don't want to use it because of this warning:

[image]

I am using EVSE with an HA integration. **I need and rely on the EVSE to be able to detect when a car is plugged.**, which is translated in mqtt with `vehicle = 0 or 1`.

[image]

So if RAPI is removed, what are the options left to still be able to work around the sleep issue but still be able to detect when a car is plugged ?

Thanks!

KipK, 2023-05-14:
There's #512 related to this issue

mathieucarbou, 2023-05-14:
> 

Oh thanks! I didn't see this one!

KipK, 2023-05-15:
I've added the evse restart endpoint to the newflags branch & ui

mathieucarbou, 2023-05-15:
> I've added the evse restart endpoint to the newflags branch & ui

@KipK : and what what about the ammeter calibration (`#SA 227 0` for me) ? Will there be a way to set this calibration from the UI if RAPI si removed ?

KipK, 2023-05-15:
yes I need to make it configurable on the UI , you can already use the /config endpoint to set the "scale" & "offset" values

cholawo, 2023-05-17:
@jeremypoulter I don't believe there's any http api to directly set the pilot current.  Is that correct?
I am concerned by the move to deprecate RAPI as it prevents me from manually correcting issues with current on the fly should [eco mode go awry](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/388#event-8452192798), or I find my night time charge is stuck at 6A (really irritating and happened last night).  I was previously able to send RAPI command when I noticed pilot was wrong, but it seems that last night my automation sent a command to RAPI pilot to 32A but it didn't seem to have an effect, perhaps RAPI lockdown has already begun!

mathieucarbou, 2023-05-17:
> I don't believe there's any http api to directly set the pilot current. Is that correct?> 

There are HTTP and MQTT APIs to set a pilot current: this is the manual override mode.

MQTT integration example: http://gist.github.com/mathieucarbou/92a3d5e0dc38d6b68aa1bdaf153a80c5

I am sending to the topic `openevse/override/set` the payload `{"state": "active", "charge_current": 16, "auto_release": true}`

KipK, 2023-05-17:
@cholawo using RAPI as you're doing currently  is messing up the wifi gateway. You can't use booth without expecting issues.
RAPI is not deprecated for those who want to communicate directly with the evse module, without running the current wifi gateway. But Wifi gw needs exclusive access on RAPI to prevent concurrent commands. You can mostly achieve everything using the API now.

cholawo, 2023-05-18:
Thanks both.
@KipK it sounds like maybe I might be making matters worse when trying to correct issues by using rapi commands!
@mathieucarbou so if at night I detect the charge timer is charging at 6A and I send the manual override mode command to get to 32A, will charging still stop at the end of the timer or does manual override prevent charge from stopping?   If during the day I were to override the current in eco mode does eco mode remain on and able to stop charge when cloudy?   My impression from the spotlight docs was that this override command is also turning charge on or off, not merely altering pilot.

mathieucarbou, 2023-05-18:
> Thanks both.
> @KipK it sounds like maybe I might be making matters worse when trying to correct issues by using rapi commands!
> @mathieucarbou so if at night I detect the charge timer is charging at [...]

Here is what I do:

When off-peak hours triggers, I am sending to the topic `openevse/override/set` the payload `{"state": "active", "charge_current": 16, "auto_release": true}`.

When peak hours trigger, I am sending to the topic `openevse/override/set` the payload `clear`. This automatically switch back to eco mode (if you were already in eco mode previously), with a pilot decided by the solar divert module.

I don't really understand your timer thing, IMO, it does not seem necessary: OpenEVSE already has a limit capability where in the UI you can stop the charge based on conditions, like charge duration, power sent to vehicle, battery capacity reached, etc

cholawo, 2023-05-18:
I'm not actually trying to do anything outside of the wifi module, I use both timer and eco mode in the wifi module and tend to leave both on all the time - to charge from sun in the day, and at night with timer.   The two issues I had are that I  that either eco mode charging was at full power (I think this might have since been fixed), and more recently that the timer at night was charging at the slowest rate.    I was merely trying to find a command that could modify the pilot value without altering eco mode and timer, to ensure pilot was correct under their conditions.   In an ideal world I wouldn't need to intervene.   I wonder if a better option rather than send RAPI when pilot is wrong, is just to restart the wifi module and hope it recovers, but maybe that is risky since the issue is happening during a charge.    
Going forwards I'll stop sending any RAPI commands and see if I still get pilot current issues, and start a new github issue when I do so it can be investigated.

KipK, 2023-05-18:
Have you updated wifi module to latest firmware? And what fw version is installed on your openevse module?

cholawo, 2023-05-18:
fw is 7.1.3.T2 and wifi is 4.1.9.  I'll update wifi to 4.2.2 now.   Thanks for all your input!

KipK, 2023-05-18:
you should update your openevse fw too if you can. There's important fix for T2 socket

cholawo, 2023-05-18:
ok, I'll try get round to that.   I just did some tests and still see my problem so I'll start a new issue for it.

jeremypoulter, 2023-07-05:
We will leave the support for `$Gx` and disable the other functions via a build option, so advanced users can re-enable support.

jeremypoulter, 2023-07-05:
Also not going to block on #512 for this ticket or the v5 release

cholawo, 2023-07-05:
Thanks @jeremypoulter.  My charger is now behaving after I stopped sending RAPI commands so I blame myself for all the unnecessary conversation in this thread

PR #702 (2023-08-30): Block all RAPI commands that are not the read only `$G...` ones
This is because the ESP32 now does a lot of the functionality and sending RAPI commands can disrupt this

Fixes #643
commit 3e586b45 (2023-08-30): Block all RAPI commands that are not the read only `$G...` ones

This is because the ESP32 now does a lot of the functionality and sending RAPI commands can disrupt this

Fixes #643
closed 2023-08-30 by glynhudson