OpenEVSE/openevse_esp32_firmware #653: OCPP Authentication key

rvoosterhout, 2023-05-22:
Hello,

Where in Steve can I find the Authentication Key? My openevse was a few versions behind, but now I need to fill in the authentication key.

Thanks.

jeremypoulter, 2023-05-22:
@matth-x can you help?

matth-x, 2023-05-22:
You can leave this field blank. SteVe doesn't support Basic Authentication (unfortunately).

The other fields can remain the same as in the previous version.

Please reopen if you have follow-up questions.

closed 2023-05-22 by matth-x

KipK, 2023-05-22:
@matth-x oups I thought it was needed. I've made the password required on UI. I need to change that then. 
When I've tested I used the admin password as key and it worked. I don't remember If I have tried putting anything :)

rvoosterhout, 2023-05-22:
Thanks for the quick fix!

pir8radio, 2023-07-21:
did this get reverted?  I have tried to connect to several OCPP servers and none connect, (none use this auth key) so im not sure if thats messing things up?

[image]

matth-x, 2023-07-23:
In my latest tests everything worked as expected. The fix is inside the submodule `gui` and I could imagine the submodule is not checked out at the latest commit.

pir8radio, 2023-07-23:
Either way I don't think this key should be "required" in future releases.