OpenEVSE/openevse_esp32_firmware #661: ESP32 wifi down after firmware update.

Silverknight87, 2023-05-25:
I recently assembled the openevse, everything was working ok. Today, i just checked for updates for the esp32 and found 4.2.2 at the top on the releases page. IIRC the wifi esp32 was on 4.1.2. I uploaded the file openevse_esp32-gateway-f_gui-v2.bin. 

i lot connectivity. on the display, i saw wifi update successful. i power down the unit and powered up and can't reach the device even after. Not sure what i did wrong. Any ideas on how to recover?

chris1howell, 2023-05-25:
ESP32 Gateway F looks like this:
[image]
Does the board you have match the firmware uploaded? If no you may need to upload the correct firmware via esptools and a USB serial adapter.

Silverknight87, 2023-05-25:
Oops, this is what i have. What do i do next. 

i see the tx,rx and gnd pins. can you point me to the 3.3v pin? The pin headers are the smaller sized ones, so its going to be a struggle. i'll try.

Also, can you please point me to the right firmware i need to use for this board?

[image]

KipK, 2023-05-25:
Use comlon openevse_wifi_v1 file for this board

chris1howell, 2023-05-25:
The Firmware is:
openevse_wifi_v1... ...bin

Power via 5v on the same pin as the red wire. The programmer in the OpenEVSE store comes with the correct JST harness.

Flashing instructions are in the user manual here:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/docs/user-guide.md#via-usb-serial-programmer

Silverknight87, 2023-05-26:
Thanks, I did manage to upload the prescribed firmware but the ap does not show up. I did try flashing with the bootloader+ partitions as well.

Is wifi supposed to come online with 5v power? 

I did try connecting it to the openevse unit as well but still nothing lighting up... once in a while the greenlight glows when i am flashing but no wifi.

KipK, 2023-05-26:
do a complete erase of the flash first. If you've used OTA before, it can still try to boot the wrong partition. 
python -m esptool --chip esp32 erase_flash

Then flash the correct fw and bootlaoder + partition

KipK, 2023-05-26:
@chris1howell I think we should change the doc to put this command instead: 

`esptool.py --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 firmware.bin 0x150000 firmware.bin`

This flash firmware.bin to booth OTA partition. When OTA switch is set to partition 2 then flashing only the first one don't solve anything.

Silverknight87, 2023-05-26:
Excellent! Thankyou @KipK - The erase flash and then reflashing seems to have done the job...  

I was aware of needing to erase from my past experiences with ESP chips but did not strike me today at all. thank you for reminding.

closed 2023-06-09 by glynhudson