OpenEVSE/openevse_esp32_firmware #662: More accurate build names

chris1howell, 2023-05-25:
Many users are uploading the wrong build for their hardware. File names for all hardware begin with openevse which is misleading. The following names are suggested to match the manufacture of hardware.

openevse_esp32-gateway-f_gui-v2.bin
olimex_esp32-gateway-f_gui-v2.bin

openevse_huzzah32.bin_gui-v2.bin
adafruit_huzzah32.bin_gui-v2.bin

KipK, 2023-05-25:
It has changed now for dev builds . UI 1 has been removed. Next release
will use common names.

KipK, 2023-05-25:
Oups I've misread, thought you were talking about the gui_v1 v2

jeremypoulter, 2023-05-25:
Yeah agreed, now we have our own OpenEVSE board these names can be misleading, will update

chris1howell, 2023-05-25:
It may also be a good idea to prefix the openevse version so it is the first on the list. It should be by far the most used version.

glynhudson, 2023-08-11:
Totally agree, good idea :+1: 

Hopefully in the future it won't be so much of an issue since GUI V2 enables updating direct from github which should choose the correct release.

PR #712 (2023-08-17): More accurate build names
Fixes #662
commit d1b68646 (2023-08-17): More accurate build names
Fixes #662
closed 2023-08-17 by chris1howell