OpenEVSE/openevse_esp32_firmware #672: Eco mode without OpenEnergyMonitor

Silverknight87, 2023-06-04:
Is having a Emonpi or emonecms etc mandatory for the Solar PV divert feature? From the openevse console, i see that openevse just makes the adjustment based on the Power value it sees on the mqtt topic configured. So, to me it feels like openevse does not care where that value is coming from right? the source could be any other tooling we might have for energy monitoring that just sends power value on the relevant mqtt topic? Is my understanding correct?

chris1howell, 2023-06-04:
Correct, it does not matter the source.  The watts available just needs to make its way to MQTT.

Here is a Python sample for Tesla Energy Gateway and Sense Energy Monitor. It grabs the data and publishes to MQTT. https://github.com/OpenEVSE/Solar_MQTT

There is also a Issue #120 to natively pull data directly from Tesla Energy Gateway, which will not require MQTT at all.

Silverknight87, 2023-06-05:
Thanks, understood.

closed 2023-06-05 by Silverknight87