OpenEVSE/openevse_esp32_firmware #676: Eco Divert set, unit starts at night

lebsd, 2023-06-11:
V.4.2.2.  My understanding is that when the mode is set Auto (let OpenEvse drcide) and Eco divert (PV power ratio set to 1.1), the unit will charge IF there is excess electricity regardless of any scheduled times.

Issue: Unit started charging at night when there was no excess and use the schedule times that should only be used when Fast (vs Eco) charge is set. 

Background: I submitted a previous issue where I had set a time to disable with Auto and Eco divert and the unit did not rurned off at specific time.   The response was Auto Eco does not use the scheduled times.

jeremypoulter, 2023-06-11:
PV divert and the schedule are independent of each other, as in enabling one does not affect the other, so if you have a schedule set for night time charging it will run regardless of the Eco / Fast mode.

lebsd, 2023-06-11:
Thanks for your quick response.   I am still confuse with the logic.  My
previous issue report said that while there was still solar, charge is not
controlled by schedule. Can you point me to the module or modules that have
the logic for the divert and also the schedule?  I have some basic
programming knowledge and would like to follow the current logic.

Perhaps the logic set initially was based on solar only, but now many are
using Batteries such as myself.  As an example last night: I set my
batteries to operate during peak hours only specifying the max load Watts
supplied by the batteries during those hours.  So I can control the rate of
discharge of my batteries based on load.  Any shortage of power to local
load is then pulled from the grid. I assumed Mode Eco meant adjusting based
on solar availability, therefore no solar, no power to charge the car
independent of what TOU. Instead charging was set to max so it drained my
battery faster.  If I wanted to charge with battery and grid during the
TOU, I would have set the toggle charge to Enabled (vs let evse decide) and
Fast.  (in the present logic, looks like setting Let Eco decide with Eco
mode behaves the same way as Enabled and Fast charging forcing me to
physically disconnect the car at night to prevent battery drain or modify
my TOU schedule. I would recommend this logic to make better use of Toggle
charge options and mode and battery users:

Toggle Charge          Mode         TOU           Action
Enabled                    Fast           not set       Charge immediately
at the max Amp rate set for the unit

Enabled                    Fast            set            Charge
immediately at the max Amp rate set for the unit AND On/Off dependent in
TOU schedule

Let Eco decide         Eco             not set      Charge based on divert
parameters amps available  (ie will only charge when solar is available, no
night charging)

Let Eco decide         Eco              set           Charge based on
divert parameters amps available AND TOU,  only charge when solar is
available, *never night charging *(if there is no start time set, it will
start when solar excess is available, ie 6 Amps min.  If start is set but
no stop, it will stop when < 6A)

Let Eco decide         Fast            not set      (same as Enabled, Fast,
TOU not set above)

Let Eco decide         Fast            set            (same as Enabled,
Fast, TOU set above)

Thanks for your efforts !
Luis

KipK, 2023-06-11:
this has changed in dev branch. Now toggling enable charge will ignore divert ( i e Eco disabled ), and Fast button has been removed, there's only an Eco toggle now.

closed 2023-06-11 by glynhudson