OpenEVSE/openevse_esp32_firmware #680: Unable to charge with current > 0.4A

ruimarinho, 2023-06-25:
Hi,

I've just updated to OpenEVSE (8.2.3.EU) and OpenEVSE Wifi v2 GUI (local_master_b8d1addc) and I can't seem to be able to charge at more than 0.4A, independently of the max current I set for the device. I'm a on a three phase system based in EU.

[image]

I turned off the self-production awareness mode and I still can't raise the current. I also can't get the car to charge for more than ~5 minutes. I don't have any automation enabled or schedules / limits set.

What could be the issue here? Thanks!

ruimarinho, 2023-06-25:
`/config`:

```
{
    "mqtt_supported_protocols": [
        "mqtt",
        "mqtts"
    ],
    "http_supported_protocols": [
        "http"
    ],
    "buildenv": "openevse_nodemcu-32s",
    "version": "local_master_b8d1addc",
    "wifi_serial": "xxx",
    "protocol": "-",
    "espinfo": "ESP32r1 2 core WiFi BLE BT",
    "espflash": 4194304,
    "firmware": "8.2.3.EU",
    "evse_serial": "6",
    "diode_check": true,
    "gfci_check": true,
    "ground_check": true,
    "relay_check": true,
    "vent_check": true,
    "temp_check": true,
    "max_current_soft": 18,
    "service": 0,
    "scale": 18,
    "offset": 8021,
    "min_current_hard": 6,
    "max_current_hard": 32,
    "ssid": "xxx",
    "pass": "_DUMMY_PASSWORD",
    "ap_ssid": "",
    "ap_pass": "",
    "lang": "",
    "www_username": "",
    "www_password": "",
    "hostname": "openevse-c160",
    "sntp_hostname": "xxx",
    "time_zone": "Europe/Lisbon|WET0WEST,M3.5.0/1,M10.5.0",
    "limit_default_type": "",
    "limit_default_value": 0,
    "emoncms_server": "https://data.openevse.com/emoncms",
    "emoncms_node": "",
    "emoncms_apikey": "",
    "emoncms_fingerprint": "",
    "mqtt_server": "xxx",
    "mqtt_port": 1883,
    "mqtt_topic": "openevse",
    "mqtt_user": "xxx",
    "mqtt_pass": "_DUMMY_PASSWORD",
    "mqtt_solar": "",
    "mqtt_grid_ie": "shellies/shellyem3-xxx/emeter/1/power",
    "mqtt_vrms": "shellies/shellyem3-xxx/emeter/0/voltage",
    "mqtt_live_pwr": "",
    "mqtt_vehicle_soc": "car/battery_level",
    "mqtt_vehicle_range": "car/battery_range",
    "mqtt_vehicle_eta": "car/time_to_fully_charged",
    "mqtt_announce_topic": "openevse/announce/c160",
    "ocpp_server": "",
    "ocpp_chargeBoxId": "",
    "ocpp_authkey": "",
    "ocpp_idtag": "",
    "ohm": "",
    "divert_type": 1,
    "divert_PV_ratio": 1.1,
    "divert_attack_smoothing_time": 20,
    "divert_decay_smoothing_time": 600,
    "divert_min_charge_time": 600,
    "current_shaper_max_pwr": 0,
    "current_shaper_smoothing_time": 60,
    "current_shaper_min_pause_time": 300,
    "current_shaper_data_maxinterval": 120,
    "vehicle_data_src": 2,
    "tesla_access_token": "",
    "tesla_refresh_token": "",
    "tesla_created_at": 0,
    "tesla_expires_in": 0,
    "tesla_vehicle_id": "",
    "rfid_storage": "",
    "scheduler_start_window": 0,
    "flags": 119537674,
    "emoncms_enabled": false,
    "mqtt_enabled": true,
    "mqtt_reject_unauthorized": true,
    "mqtt_retained": false,
    "ohm_enabled": false,
    "sntp_enabled": true,
    "tesla_enabled": false,
    "divert_enabled": false,
    "current_shaper_enabled": false,
    "pause_uses_disabled": false,
    "mqtt_vehicle_range_miles": false,
    "ocpp_enabled": false,
    "ocpp_auth_auto": false,
    "ocpp_auth_offline": false,
    "ocpp_suspend_evse": false,
    "ocpp_energize_plug": false,
    "rfid_enabled": false,
    "factory_write_lock": true,
    "is_threephase": true,
    "wizard_passed": true,
    "default_state": true,
    "mqtt_protocol": "mqtt",
    "charge_mode": "fast"
}
```

Not sure if related, but is `service` supposed to be 0?

ruimarinho, 2023-06-26:
@KipK I am also getting weird messages on my display such as "Waiting for \n Time 10:00:05 PM". It coincides with my off-peak schedule but I've never set it on OpenEVSE - it has always been a Home Assistant automation.

glynhudson, 2023-07-05:
Are you still having this issue? Could it be a measurement issue? Could the car actually be charging at full speed?

ruimarinho, 2023-07-05:
Yes, in addition to seeing Service Level 0. Is that even possible? I thought it could only be 1 or 2.

ruimarinho, 2023-07-14:
@glynhudson After performing a full ESP32 erase, the service level is now 1 (in the UI at least). I have no explanation for how it got to level 0.

[image]

The device remains with limited capacity.

[image]

I think there must be something related to the production meter that is affecting the charging current. Please note this is a manual override to top-up whatever wasn't charged during PV divert.

`/target`:

```
{"properties":{"state":"active","auto_release":false},"claims":{"state":65537}}
```

`/override`:

```
{"state":"active","auto_release":true}
```

`/config`:

```
{
    "max_current_soft": 16,
    "service": 0,
    "scale": 16,
    "offset": 8021,
    "min_current_hard": 6,
    "max_current_hard": 32,
    …
    "divert_type": 1,
    "divert_PV_ratio": 1.1,
    "divert_attack_smoothing_time": 20,
    "divert_decay_smoothing_time": 600,
    "divert_min_charge_time": 600,
    "current_shaper_max_pwr": 0,
    "current_shaper_smoothing_time": 60,
    "current_shaper_min_pause_time": 300,
    "current_shaper_data_maxinterval": 120,
    "scheduler_start_window": 0,
    "flags": 52437514,
    …
    "divert_enabled": true,
    "current_shaper_enabled": false,
    …
    "is_threephase": true,
    "default_state": false,
    "charge_mode": "fast"
}
```

ruimarinho, 2023-07-14:
I think I know where the service level 0 is coming from - probably from miscalculation of true RMS via the configured mqtt topic?

[image]

This is what I see on the Home Assistant side. I've just removed the voltage topic and the service level immediately changed to 2, but current is still limited to 5.3A for some reason. /cc @KipK

ruimarinho, 2023-07-29:
@KipK this is still happening frequently which makes it impossible to get a full charge when needed - any chance you could devote some time understanding why the set point (max current) is jumping around?

closed 2026-09-21 by github-actions[bot]