OpenEVSE/openevse_esp32_firmware #681: Best practice to change configs programmatically?

matth-x, 2023-06-27:
Hi, I'm working on an upgrade of the OCPP module and a general clean up. Currently the synchronization between the OpenEVSE configs and the OCPP lib still doesn't work very smoothly. This is an example where I need it:

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/b8d1addc4fcf47c51b1c559e5a10d9c0f382bfff/src/ocpp.cpp#L86-L91

The OCPP server can change the WebSocket URL to allow remote configuration of the charger. The new URL enters the OpenEVSE configs store here, but the change is not applied in the web model. I cannot say for sure, but I think this means that the web model potentially overwrites the updates later with its old data.

Is there anything I should take extra care of when doing this or a better way to update the configs?

jeremypoulter, 2023-06-27:
That should work, all goes to the same store... Possibly the older version did not update the config version number so the changes would not be pulled to an already loaded web UI but this should be fixed in the latest master code.

matth-x, 2023-06-28:
Probably I'm doing the two-way sync wrong, but I have problems understanding the reason here. To track the issue down, I set the OCPP task up after erasing the flash and got the following debug output:

```
 ** Fresh start with empty flash, OCPP backend set via UI, now enable OCPP task **

config_ver = 7
{"config_version":7}
Saving config
Writing {"ws":"***","wp":"***","ows":"ws://***/steve/websocket/CentralSystemService","cid":"connection-test","f":35651592} to EEPROM
157 bytes written to EEPROM @ 0x0000, committing
Done
ocpp_enabled changed

 ** execute OCPP_initialize **

[AO] debug (ConfigurationContainerFlash.cpp:30): Populate FS: create configuration file
[...]
[AO] debug (ConfigurationKeyValue.cpp:129): Initialized config object:
[AO]     > Key = AO_FreeVendActive, value = true
[AO] debug (ConfigurationKeyValue.cpp:287): Initialized config object:
[AO]     > Key = AO_FreeVendIdTag, value = DefaultIdTag
[...]
```
Until here, the OCPP lib is initialized. The config `AO_FreeVendActive` which corresponds to `config_ocpp_auto_authorization()` is true per factory default.

This snippet follows next, after completing `OCPP_initialize` when `config_commit()` is executed with the OCPP factory defaults:

```
 ** OCPP_initialize finished, now update OpenEVSE config store and execute config_commit() **

[ocpp] Sending new configs:
       ocppSocket->getBackendUrl(): ws://***/steve/websocket/CentralSystemService 
       ocppSocket->getChargeBoxId(): connection-test
       ocppSocket->getAuthKey():
       freevendActive: true
       freevendIdTag: DefaultIdTag 
       allowOfflineTxForUnknownId: true
[ocpp] resulting JSON: {"ocpp_server":"ws://***/steve/websocket/CentralSystemService","ocpp_chargeBoxId":"connection-test","ocpp_authkey":"","ocpp_auth_auto":1,"ocpp_idtag":"DefaultIdTag","ocpp_auth_offline":1}
ocpp_idtag changed
[ocpp] Receiving new configs:
       ocpp_server.c_str(): ws://***/steve/websocket/CentralSystemService 
       ocpp_chargeBoxId.c_str(): connection-test
       ocpp_authkey.c_str():
       config_ocpp_auto_authorization(): false 
       ocpp_idtag.c_str(): DefaultIdTag
       config_ocpp_offline_authorization(): false
[...]
ocpp_auth_auto changed
[ocpp] Receiving new configs:
       ocpp_server.c_str(): ws://***/steve/websocket/CentralSystemService
       ocpp_chargeBoxId.c_str(): connection-test
       ocpp_authkey.c_str():  
       config_ocpp_auto_authorization(): true
       ocpp_idtag.c_str(): DefaultIdTag
       config_ocpp_offline_authorization(): false
[...]
ocpp_auth_offline changed
[ocpp] Receiving new configs:
       ocpp_server.c_str(): ws://***/steve/websocket/CentralSystemService 
       ocpp_chargeBoxId.c_str(): connection-test
       ocpp_authkey.c_str():
       config_ocpp_auto_authorization(): true 
       ocpp_idtag.c_str(): DefaultIdTag
       config_ocpp_offline_authorization(): true
Flushing RAPI command queue ... Done
config_ver = 8
{"config_version":8}
Saving config
Writing {"ws":"***","wp":"***","ows":"ws://***/steve/websocket/CentralSystemService","cid":"connection-test","idt":"DefaultIdTag","f":48250888} to EEPROM
178 bytes written to EEPROM @ 0x0000, committing
Done

 ** after config_commit() **
```

The initialization stage is completed here and the last received config notification is correct. In the following loop() stage, the string configs remain stable, but the configs based on the flags are reverted:

```
[...]

 ** enter loop() stage **

ocpp_auth_auto changed
[ocpp] Receiving new configs:
       ocpp_server.c_str(): ws://***/steve/websocket/CentralSystemService 
       ocpp_chargeBoxId.c_str(): connection-test
       ocpp_authkey.c_str():
       config_ocpp_auto_authorization(): false
       ocpp_idtag.c_str(): DefaultIdTag 
       config_ocpp_offline_authorization(): true
[...]
ocpp_auth_offline changed
[ocpp] Receiving new configs:
       ocpp_server.c_str(): ws://***/steve/websocket/CentralSystemService
       ocpp_chargeBoxId.c_str(): connection-test 
       ocpp_authkey.c_str():
       config_ocpp_auto_authorization(): false
       ocpp_idtag.c_str(): DefaultIdTag
       config_ocpp_offline_authorization(): false
Flushing RAPI command queue ... Done
config_ver = 9
{"config_version":9}
Saving config
Writing {"ws":"***","wp":"***","ows":"ws://***/steve/websocket/CentralSystemService","cid":"connection-test","idt":"DefaultIdTag","f":35667976} to EEPROM
178 bytes written to EEPROM @ 0x0000, committing
Done
[...]

 ** OCPP waiting to be connected, no further config changes **
```

Do you see a good reason for the sync failure here?

I will set up a draft PR with the code producing this output.

matth-x, 2023-07-05:
I was able to figure out what the difficulty was. When pressing the enable button in the GUI, the apply configs callback in the OCPP task sent back a new configs update with the OCPP factory defaults at the same time. I think that the factory defaults should be defined in the `app_config` module anyway because they are more related to the OpenEVSE configs store. So I conclude that this was the "bad practice" here.

closed 2023-07-05 by matth-x