OpenEVSE/openevse_esp32_firmware #683: solar production management

monfiston, 2023-06-28:
Good morning
I note that when using solar mode, the evse terminal changes the intensity values, but there are always around 300/400w which are reinjected into the electricity network instead of going into the car

here is my setup
[export.json.zip](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/files/11897929/export.json.zip)
[image]

here is an example, but I have many others
in blue the consumption curve of my house which is positive when I consume current, and negative when I send current to the national electricity grid
and in orange the openevse intensity setpoint
[image]
as you can see for almost 1 hour, I sent back to the electrical network 300w on average constantly

here is the same graph all day
[image]

closed 2023-07-04 by monfiston

KipK, 2023-07-04:
@monfiston is the issue solved on your side?

monfiston, 2023-07-04:
Non, mais je pense que le pb provient de mon interface openevse(copie de mauvaise qualité)