OpenEVSE/openevse_esp32_firmware #684: Shaper - Rapid toggling of EVSE in and out of sleep for <6A available faults vehicle charging system

DrFrankReade, 2023-06-28:
When the load shaper has less than 6A available for charging, the charger sleeps, pausing the charge.  THIS BEHAVIOR MAKES SENSE and is the way it ought to work. 

However, with a rapidly cycling load, like an electric oven or air conditioner, the load shaper faults my Mach-E after rapidly entering and leaving the sleep state after a certain number of cycles or some kind of rate limit. 

I don't know the best way to to implement a change in the UI, such that either the EVSE never sleeps and only ever goes to 6A (acceptable over-limit), OR the EVSE can go to sleep, but won't wake up until 6A has been available for at least x minutes (a rate limit) 

I can say that the Mach-E doen't mind constantly changing current, as I implemented my own shaper with HomeAssistant, but the on-off-on-off-on-off-on-off  stuff makes the car get pretty angry.

And besides, the built-in shaper actually works really well most of the time. It's just come to a head now that we're entering summer and the air conditioner is starting to run and really give the algorithm a workout.

glynhudson, 2023-06-29:
What have you got the 'minimum charge time' setting to? The default is 600s which is 10min. Stopping and then starting 10min later should not be too fast, but you can increase this parameter if you want. 

What do you mean by 'faults the charging system'? Does the car display an error? 

[image]

DrFrankReade, 2023-07-01:
Mustang Mach E "car" shows an error on the instrument cluster and app as a charging station fault, and attempts to charge end until the J1772 plug is physically cycled. 

To reiterate, there's no solar configured, just a whole-house MQTT power topic coming in to the shaper to limit overall current draw. 

 All settings in the filters for PV / divert are  set to default, and I had been ignoring them, assuming they didn't apply.

I've been thinking about the UI, maybe a checkbox that say "Enforce minimum charge current" or "Prevent sleep"
with a short explainer / tool tip "Prevents EVSE from entering sleep mode when less than 6A is available, shaper current limits may be exceeded by 6A"

This sort of reminds me of the weirdness of the VW e-Golf - It hated sleep mode as well and would freak out and not charge. But it wasn't technically J1772 (or whatever) compliant. I'm not sure that the Mach-E is either, but who knows.

DrFrankReade, 2023-07-05:
[image]
This is the consistent result of the shaper sleeping the EVSE.

glynhudson, 2023-07-05:
Ah right, can you try adjusting the Shaper parameter e.g increasing the minimum pause time?

[image]

DrFrankReade, 2023-07-05:
That UI doesn't seem to be visible in 4.2.2 - is this the beta?
I just get the max power and the topic. OpenEVSE fw is 7.1.3

DrFrankReade, 2023-07-17:
[image]
The EVSE now has FW 8.2.3 and the ESP32 had both 4.2.2 and V2 GUI pre-release 2023-05-10T05:28:39Z
[image]
Neither one of these has shown the shaper GUI you've shared with us above, just the basic one above.

This is the NON-EU builds everywhere, not sure how this would matter unless it does. 
What do you suppose is going on?

Meanwhile, I've changed the PAUSE STATUS from **Sleep** to **Disable** and we'll see what happens there. 
[image]
I know that the e-Golf didn't like sleep mode on the OpenEVSE.

DrFrankReade, 2023-08-21:
This is what I did to make the charger set to 6A instead of pausing - Somebody tell me if this was dumb or not... 
`unsigned long CurrentShaperTask::loop(MicroTasks::WakeReason reason) {

	if (_enabled) {
			EvseProperties props;
			if (_changed) {
				props.setMaxCurrent(floor(_max_cur));
				if (_max_cur < evse.getMinCurrent()) {
					// pause temporary, not enough amps available
					//props.setState(EvseState::Disabled);
					props.setMaxCurrent(6); // Set the charging current to 6 amps instead of pausing, because pausing pisses off the Mach-E
					if (!_pause_timer)
					{
						_pause_timer = millis();
					}

				}
				else if (millis() - _pause_timer >= current_shaper_min_pause_time * 1000 && (_max_cur - evse.getMinCurrent() >= EVSE_SHAPER_HYSTERESIS))
				{
					_pause_timer = 0;
					props.setState(EvseState::None);
				}
				_timer = millis();
				_changed = false;
				// claim only if we have change
				if (evse.getState() != props.getState() || evse.getChargeCurrent() != props.getChargeCurrent())
				{
					evse.claim(EvseClient_OpenEVSE_Shaper, EvseManager_Priority_Safety, props);
					StaticJsonDocument<128> event;
					event["shaper"] = 1;
					event["shaper_live_pwr"] = _live_pwr;
					event["shaper_smoothed_live_pwr"] = _smoothed_live_pwr;
					event["shaper_max_pwr"] = _max_pwr;
					event["shaper_cur"] = _max_cur;
					event["shaper_updated"] = _updated;
					event_send(event);
				}
			}
			else if ( !_updated || millis() - _timer > current_shaper_data_maxinterval * 1000 )
			{
				//available power has not been updated since EVSE_SHAPER_FAILSAFE_TIME, pause charge
				DBUGF("shaper_live_pwr has not been updated in time, pausing charge");

				if (_updated)
				{
					_pause_timer = millis();
					_updated = false;
					_smoothed_live_pwr = _live_pwr;
				}

				if (evse.getState(EvseClient_OpenEVSE_Shaper) != EvseState::Disabled)
				{
					//props.setState(EvseState::Disabled);
					props.setMaxCurrent(6); // Again, set the current to 6 instead of disabling. More of a fail-charging 
					evse.claim(EvseClient_OpenEVSE_Shaper, EvseManager_Priority_Limit, props);
					StaticJsonDocument<128> event;
					event["shaper"] = 1;
					event["shaper_live_pwr"] = _live_pwr;
					event["shaper_smoothed_live_pwr"] = _smoothed_live_pwr;
					event["shaper_max_pwr"] = _max_pwr;
					event["shaper_cur"] = _max_cur;
					event["shaper_updated"] = _updated;
					event_send(event);
				}
			}
	}`

KlugFR, 2024-01-31:
The "rapid toggling" seems to be an issue also when using solar divert and clouds suddenly appear.

It looks like the charger stops sending power to the car _before_ telling it to go in pause mode.
This causes the car to think there's a charger issue, thus refusing charge to resume when clouds go away.

With a BMW i3 I have the same behavior than your Mustang (I have to physically disconnect/reconnect the plug).
With an Hyunday Ioniq 5, I can use the car app on my phone to stop than start the charge.

Some screenshots here: https://community.openenergymonitor.org/t/pv-divert-and-ioniq-5-ev6

closed 2026-09-21 by github-actions[bot]