OpenEVSE/openevse_esp32_firmware #686: Shaper should support Multiple circuits for decsion making.

ausil, 2023-07-01:
This may well be an oddity of how electricity in the US works, but I have two wattages that I need to consider when configuring shaper.

My house has a 200A service, and there is a 125A circuit running from it to the garage. Currently, there are 2 OpenEVSE's in the garage, but sometime in the next year or so, I plan to add a 3rd unit from a circuit on the main panel. Both units in the garage are 48A units. I would like to have a wattage for the garage usage to ensure I do not overload the sub panel in the garage, but I also need to check the overall house usage and ensure that I am not overloading it. I have an emonpi monitoring the house usage and would need to look at adding some extra monitoring for the sub panel. But with that in place, it seems reasonable to ensure that neither load capacity is exceeded.

chris1howell, 2023-07-02:
The Emonpi has Node-Red installed. Node-Red could easily handle the custom
logic and send the result to shaper.

closed 2023-07-05 by glynhudson

ausil, 2023-07-11:
I would like to know what I am actually supposed to do here, because neither closing nor suggesting node-red are helpful