OpenEVSE/openevse_esp32_firmware #690: "Quit Wizard" button does not accomplish anything

fhteagle, 2023-07-05:
Just loaded the  b8d1add  dev release on one of my units. After the firmware update completed, the unit restarted and the web UI went to the config wizard. Some of the config carried forward correctly, but the time went from NTP setting on old firmware to manual on new firmware. No big deal, fixed that, and hit next until the last screen where the button changed to "Quit Wizard". Clicked that, and the UI just went back around to the first card of the wizard, not to the main UI. Sat and waited maybe 30 seconds to see if it would reload itself or something, but no, just stayed on that card. Cleared the extra bits off the URL and manually went to the main UI no problem.

So the "Quit Wizard" button needs to go to the main URL for the unit, not back to the first config card.

KipK, 2023-07-05:
Have you also updated and build the gui first?

It displays the Wizard only if the wizard_passed flag is not set.

fhteagle, 2023-07-05:
Downloaded the precompiled .bin from here, not from a local compile.

ruimarinho, 2023-07-07:
Similar thing here. In my case it required going through the wizard twice.

DrFrankReade, 2023-08-21:
Same here, cloned and built from source  on August 20, 2023. 
[image]

cscott, 2023-09-11:
Same here, twice -- ie, on both my chargers, I got sent back to the beginning by quit wizard the first time and had to run the wizard twice.  Maybe a race condition?  (Ie, the first page of the wizard is loaded before wizard_passed has been set?)

closed 2023-09-13 by glynhudson