OpenEVSE/openevse_esp32_firmware #691: Divert mode active while importing energy from the grid

ruimarinho, 2023-07-07:
Hi!

I've recently updated to v2 GUI along with 8.2.3.EU and I've been experiencing several new bugs. This one is related to divert mode - it can become active even when importing energy from the grid.

[image]

In this example, I have an MQTT export topic indicating 1920W being imported from the grid, yet OpenEVSE still activates divert mode while in eco mode if I switch to "auto".

[image]

I don't think this is expected - or am I misunderstanding this feature? Thanks for the help!

dakaix, 2023-08-16:
I am also experiencing a similar issue, likely relating to the same bug. In my case OpenEVSE fails to pause the charge after an extended period at 0A. In my environment I have Home Assistant calculate a value for excess PV capacity (in W), based on other controlled device loads within the house. 

However once this drops to 0 (because the battery SoC fell below the threshold, or a lack of PV), then the charging session doesn't stop. Rather it continues at 6A until I intervene by stopping the charge via the Tesla app.

So far it's happened twice in the last two days, which is a 100% record for when I enabled the Self Production divert! :)

**Screenshots**
OpenEVSE dashboard, note the "Last Updated" here was in excess of 3 hours until I tried manually re-pushing a value of "0" to MQTT - this didn't trigger Divert to pause, even after the subsequent 600 second timer had expired.
[image]

OpenEVSE & Sensor Data, note the state change from "Charging" to "Connected" was a result of my stopping the charge via the car manually... and that the charge didn't pause between 14:36 and 15:12 even though the production value was 0.
[image]

Sensor History:
[image]

OpenEVSE Current State Changes:
[image]

jeremypoulter, 2023-08-16:
Can you attach a CSV of the data feed?

dakaix, 2023-08-16:
> Can you attach a CSV of the data feed?
Sure, PFA.

[entities-2023-08-16_23_09_48.csv](https://github.com/OpenEVSE/openevse_esp32_firmware/files/12363942/entities-2023-08-16_23_09_48.csv)

dakaix, 2023-08-17:
Issue reoccurred again this afternoon... screenshots & data below again. It did manage to stop 3-4 times during the afternoon, but not at the end of the day!

I've also included screenshots of the HA automation traces the last two executions, which shows the last two values posted on MQTT.

[image]

[image]

[image]

[image]

[image]

[entities-2023-08-17_17_17_05.csv](https://github.com/OpenEVSE/openevse_esp32_firmware/files/12371532/entities-2023-08-17_17_17_05.csv)

400bird, 2024-02-11:
Any update on this? I am also seeing the same issue.
I was previously on Wifi firmware 4.1.0 and hardware firmware 7.1.3 (? I thought I documented that but I guess not)
Solar Divert using MQTT worked fine there.

Today, I updated to 8.2.2 and 5.0.2 (US)
I can see both production and smoothed current updating in the UI, but the EVSE is ignoring that and charging at full power.

Looks like there is also at least one other report of this issue:
https://github.com/OpenEVSE/openevse_esp32_firmware/issues/692#issue-1794385483

400bird, 2024-03-16:
Just to follow up, my issue was that the timer overrides eco mode. Once I deleted the timers, eco mode functions normally.

https://github.com/OpenEVSE/openevse_esp32_firmware/issues/812

closed 2026-09-21 by github-actions[bot]