OpenEVSE/openevse_esp32_firmware #696: Shaper above max allowed power

gonzalezcalleja, 2023-07-18:
Hi,

I'm using shaper with last dev build (master_517522f7) with this configuration:

[image]

Live powered from shelly, updated every 3 seconds.

But the power used is above 5700:

[image]

In some cases it has exceeded 7500 and has caused a supply cut.

KipK, 2023-07-18:
Can you post the published values from your Shelly?
What is your max current setting btw? It should not be above your 5700w
limit then, 23A.

gonzalezcalleja, 2023-07-18:
> Can you post the published values from your Shelly? What is your max current setting btw? It should not be above your 5700w limit then, 23A. Le mar. 18 juil. 2023, 11:22, Javier González Calleja < ***@***.***> a écrit :
> […](#)

Hi @KipK, thanks for your answer.

About the values from the shelly, I think they are correct because I was using shaped with last version without problem until this update to dev version.

My max current settings is 32A, but with shaper I think that the first limit should limit by the "max power allow".

gonzalezcalleja, 2023-07-19:
From mosquitto_sub, voltage:

[image]

Power:
[image]

Log from openevse:

[image]

graememk, 2023-08-11:
Running the latest dev build, I'm coming across this too.

I'm using it to limit to the output of the batteries, but it didn't throttle the charge last night at all

Rolling back to the dev pre-release (may 15) it works as expected.

MilorES, 2024-01-28:
OpenEVSE: 8.2.0.EU
OpenEVSE Wifi: v5.0.2

Same problem here:
[image]

How can you say that you have free 11.65A, 2800 - 1679 = 1130 / 230 = 4.9A

What version is recommended?
What is the decimal limit on voltage and power?
How long do you have to wait between data? Is 1s good? 5s? 10s?

I had to limit it to 6A, which is the minimum so as not to run out of electricity.

graememk, 2024-01-28:
@jeremypoulter could this be merged with this one and closed? https://github.com/OpenEVSE/openevse_esp32_firmware/issues/762

@MilorES this is fixed in the nightly builds agaik and should be in the next release

jeremypoulter, 2024-01-28:
Yeah, think that is the same issue

closed 2024-01-28 by jeremypoulter

MilorES, 2024-01-29:
@graememk So is it recommended to wear a  nightly build? Or better to wait for the new version?
How can you know when it will be released?

The current one is a couple of months old.