OpenEVSE/openevse_esp32_firmware #703: OCPP MeterValues are in kWh, not Wh

serge005, 2023-07-24:
I'm on 4.2.2 and noticed that MeterValues reported via OCPP are in kWh, not Wh. Here's a sample of meter values received for one of my charge sessions. You can see that it took 7 minutes to increment 1kWh (361 to 362) while charging at 9400W.

meterValues: [
  [
    {
      timestamp: '2023-07-01T21:45:30.000Z',
      sampledValue: [
        {
          value: '9451',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:46:30.000Z',
      sampledValue: [
        {
          value: '9451',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:47:30.000Z',
      sampledValue: [
        {
          value: '9398',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:48:30.000Z',
      sampledValue: [
        {
          value: '9451',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:49:30.000Z',
      sampledValue: [
        {
          value: '9398',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:50:30.000Z',
      sampledValue: [
        {
          value: '9398',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '361',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
  [
    {
      timestamp: '2023-07-01T21:51:30.000Z',
      sampledValue: [
        {
          value: '9398',
          context: 'Sample.Periodic',
          measurand: 'Power.Active.Import',
          unit: 'W'
        },
        {
          value: '362',
          context: 'Sample.Periodic',
          measurand: 'Energy.Active.Import.Register',
          unit: 'Wh'
        }
      ]
    }
  ],
]

serge005, 2023-07-26:
@matth-x, thanks for the quick response. This fix should work, but we will miss out on having Wh precision. Instead of multiplying kWh by 1000, can we instead get and publish the orignal Wh value before it was converted to kWh?

matth-x, 2023-07-28:
The OpenEVSE platform provides the total energy as floating point number and should be precise enough so that the multiplication by 1000 will give a reasonable Wh value. I think it is worth a try. Sorry that I cannot verify this fix on my own, I'm running the tests of the OpenEVSE without real load (which is why this error made it into the code in the first place).

serge005, 2023-07-28:
I see your point, but all meterValues I have received so far are whole numbers. There are no decimals. That means the kWh number is being rounded somewhere. Is the rounding happening further upstream from your code change? If so, then your fix should would. 

Also, on a related topic, StartTransaction and StopTransaction also report kWh. Looking at the spec for OCPP 1.6, it states the following for Start and StopTransaction.req
_meterStart, integer, Required. This contains the meter value in Wh for the connector at end of the transaction._
_meterStop, integer, Required. This contains the meter value in Wh for the connector at end of the transaction._

I would test this out for you, but it's quite a bit of effort for me to do so with load as well. Once we're confident in the changes, I'll give it a shot. Appreciate your efforts!

matth-x, 2023-07-31:
Yes, the rounding happened after the number in ocpp.cpp has been taken. And AFAIK the OCPP integration uses the same OpenEVSE endpoints for retrieving the energy measurements as the GUI where they are displayed as floating point numbers.

The Start- and StopTransaction energy values are also affected by this unit conversion.

closed 2023-09-13 by matth-x