OpenEVSE/openevse_esp32_firmware #704: Max current slider not able to adjust

js1tr3, 2023-07-25:
On manual charge mode I can't adjust the slider to set the current. Using the new wifi firmware latest github release.

KipK, 2023-07-26:
Is it locked to 6A? Check you have set the max current in settings first.

js1tr3, 2023-07-26:
It's locked to the max current... which is 35A. I can uccesfully change the
max current in the settings. Limiting is working in eco mode based on solar
production (inflow) from my emoncms server. But overnight I like to
throttle back the charging rate to achieve full charge just prior to
departure. I can use the departure setting from my car, but this worked in
the previous versions.

KristianDebrenti, 2023-10-10:
I confirm the issue. The Charge Rate slider cannot be adjusted.

Mathopenevse, 2023-11-06:
I have thé same problem with a D1 mini esp32 and firmware  adrafruit huzzah 32... did you find the solution?

danpadure, 2024-09-10:
It used to work by disabling shaper. But now it won't work at all.  All I can do is change max current in setup.
How can this be an issue only for some?
[image]

danpadure, 2024-09-10:
ok, I figured it out.  It's a programming error:
If ECO mode is ENABLED and you press the lightning button to manually start charging, the ECO mode auto-disables, BUT you can NOT adjust the charge rate
But IF you DISABLE ECO mode BEFORE pressing the lightning button to manually start charging, THEN you CAN adjust the charge rate
[image]

danpadure, 2024-09-10:
so use this workaround until someone fixes the programming issue

closed 2026-09-21 by github-actions[bot]