OpenEVSE/openevse_esp32_firmware #708: cannot compile

aaronwesel, 2023-08-02:
Am I the only one having this issue? Wihtout any modification, if I download the Wifi files from github and try to compile it using pio I get an error "Littlefs" not found. What am I doing wrong?

jeremypoulter, 2023-08-02:
What enviroment are you using to compile? Can you show a full log? We do build via a GitHub workflow [image] and that is passing.

aaronwesel, 2023-08-02:
I am using vscode pio, I followed all the instructions. I am not sure where to get these logs. Here is  screenshot if that helps.
[image]
[image]

jeremypoulter, 2023-08-02:
Which board/env are you building? Are they actually compilation errors from Platform IO, or just VSCode trying to parse the code? The latter may just be a VS Code miss configuration of VS Code and not impact building/uploading/etc

aaronwesel, 2023-08-03:
Yes, these are build errors. It is the same for every env I try. Here is
the terminal output.

 *  Executing task: C:\Users\techo\.platformio\penv\Scripts\platformio.exe
debug

Processing openevse_esp32-gateway-e_dev (board: esp32-gateway; platform:
***@***.***; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose
mode can be enabled via `-v, --verbose` option
Firmware Revision: -D BUILD_TAG=local_master_ff55cf44_modified -D
BUILD_HASH=ff55cf44_modified
CONFIGURATION:
https://docs.platformio.org/page/boards/espressif32/esp32-gateway.html
PLATFORM: Espressif 32 (6.0.1) > OLIMEX ESP32-GATEWAY
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog,
iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h,
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.20.1
|-- ArduinoMongoose @ 0.0.18
|-- Micro Debug @ 0.0.5
|-- ConfigJson @ 0.0.5
|-- OpenEVSE @ 0.0.14
|-- ESPAL @ 0.0.3
|-- StreamSpy @ 0.0.1
|-- MicroTasks @ 0.0.3
|-- ArduinoOcpp @ 0.2.0
|-- ArduinoOcppMongoose @ 0.0.1
|-- EEPROM @ 2.0.0
|-- LittleFS_esp32 @ 1.0.5
|-- Update @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Ethernet @ 2.0.0
|-- FS @ 2.0.0
|-- src @ 0.0.0+20210705113105
Building in debug mode
6.14.10
***@***.***;C:\Users\techo\.platformio\penv\Scripts;C:\Users\techo\.platformio\penv;C:\Users\techo\.platformio\python3;C:\Python39\Scripts\;C:\Python39\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program
Files\Common Files\Autodesk Shared\;C:\Program Files\Microsoft SQL
Server\130\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files\Microsoft
SQL
Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program
Files\PuTTY\;C:\Users\techo\Downloads\php-8.0.0-nts-Win32-vs16-x64;C:\ProgramData\ComposerSetup\bin;C:\ProgramData\chocolatey\bin;C:\Program
Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program
Files\Git\cmd;C:\Users\techo\Documents\GitHub\MinGW64;C:\cygnus\cygwin-b20\H-i586-cygwin32\bin;C:\Program
Files\CMake\bin;C:\Users\techo\AppData\Local\Programs\Python\Python311;C:\Users\techo\AppData\Local\Programs\Python\Python311\Scripts;C:\Program
Files\nodejs\;C:\Users\techo\AppData\Local\Microsoft\WindowsApps;C:\Users\techo\.dotnet\tools;C:\Users\techo\AppData\Roaming\Composer\vendor\bin;C:\Users\techo\AppData\Local\Programs\Microsoft
VS
Code\bin;C:\Users\techo\AppData\Local\GitHubDesktop\bin;C:\Users\techo\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;";C:\Users\techo\Documents\GitHub\esp-idf\tools";C:\Users\techo\AppData\Roaming\npm
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\app_config.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\energy_meter.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\event_log.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\main.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\manual.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\mqtt.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\net_manager.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\src\ocpp.cpp.o
[... 59 lines cut ...]
src/app_config.cpp:569:3: error: 'LittleFS' was not declared in this scope
   LittleFS.format();
   ^~~~~~~~
src/app_config.cpp:569:3: note: suggested alternative: 'bitClear'
   LittleFS.format();
   ^~~~~~~~
   bitClear
*** [.pio\build\openevse_esp32-gateway-e_dev\src\main.cpp.o] Error 1
*** [.pio\build\openevse_esp32-gateway-e_dev\src\app_config.cpp.o] Error 1
======================================================================================
[FAILED] Took 26.53 seconds
======================================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
openevse_esp32-gateway-e_dev  FAILED    00:00:26.528
=================================================================================
1 failed, 0 succeeded in 00:00:26.528
=================================================================================

 *  The terminal process
"C:\Users\techo\.platformio\penv\Scripts\platformio.exe 'debug'" terminated
with exit code: 1.
 *  Terminal will be reused by tasks, press any key to close it.
ᐧ

jeremypoulter, 2023-08-03:
Looks like you are using Windows, so could be an issue with that. Will take a look. You could also try installing Ubuntu via WSL and building via that?

matth-x, 2023-08-03:
I had the same issue but assumed that it was due to my local PIO installation (also Windows). As a workaround, I added the following line into platformio.ini (e.g. line 50).

```cpp
  -D LittleFS=LITTLEFS
```

jeremypoulter, 2023-08-03:
IIRC `LITTLEFS` was the older version of LittleFS that was supplied as an external library. LittleFS is now included as part of the main ESP32 core and should use `LittleFS`

aaronwesel, 2023-08-03:
> 

Thank you! This was very helpful and resoled this issue. Thank you Jeremy and Matthew.

jeremypoulter, 2023-08-03:
Let's keep this open till we have a proper solution for building on Windows

aaronwesel, 2023-08-03:
I am still having other issues with the code compilation, a different issue. Should we discuss it here, or not?

In file included from C:/Users/techo/.platformio/lib/src/Wire.cpp:32:
C:/Users/techo/.platformio/lib/src/Wire.h:44:5: error: 'i2c_t' does not name a type; did you mean 'ino_t'?
     i2c_t * i2c;
     ^~~~~
     ino_t
C:/Users/techo/.platformio/lib/src/Wire.h:64:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t last_error; // @stickBreaker from esp32-hal-i2c.h
     ^~~~~~~~~
     esp_err_t
C:/Users/techo/.platformio/lib/src/Wire.h:87:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t writeTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true);
     ^~~~~~~~~
     esp_err_t
C:/Users/techo/.platformio/lib/src/Wire.h:88:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t readTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true, uint32_t *readCount=NULL);
     ^~~~~~~~~
     esp_err_t
Compiling .pio\build\openevse_esp32-gateway-e_dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\openevse_esp32-gateway-e_dev\FrameworkArduino\Print.cpp.o
C:/Users/techo/.platformio/lib/src/Wire.cpp: In constructor 'TwoWire::TwoWire(uint8_t)':
C:/Users/techo/.platformio/lib/src/Wire.cpp:39:6: error: class 'TwoWire' does not have any field named 'i2c'
     ,i2c(NULL)
      ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:48:6: error: class 'TwoWire' does not have any field named 'last_error'
     ,last_error(I2C_ERROR_OK)
      ^~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:48:17: error: 'I2C_ERROR_OK' was not declared in this scope
     ,last_error(I2C_ERROR_OK)
                 ^~~~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp: In destructor 'virtual TwoWire::~TwoWire()':
C:/Users/techo/.platformio/lib/src/Wire.cpp:55:8: error: 'i2c' was not declared in this scope
     if(i2c) {
        ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:56:9: error: 'i2cRelease' was not declared in this scope
         i2cRelease(i2c);
         ^~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:56:9: note: suggested alternative: 'i2cRead'
         i2cRelease(i2c);
         ^~~~~~~~~~
         i2cRead
C:/Users/techo/.platformio/lib/src/Wire.cpp: In member function 'bool TwoWire::setPins(int, int)':
C:/Users/techo/.platformio/lib/src/Wire.cpp:63:8: error: 'i2c' was not declared in this scope
     if(i2c) {
        ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp: In member function 'bool TwoWire::begin(int, int, uint32_t)':
C:/Users/techo/.platformio/lib/src/Wire.cpp:110:5: error: 'i2c' was not declared in this scope
     i2c = i2cInit(num, sda, scl, frequency);
     ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp: In member function 'void TwoWire::setClock(uint32_t)':
C:/Users/techo/.platformio/lib/src/Wire.cpp:138:21: error: 'i2c' was not declared in this scope
     i2cSetFrequency(i2c, frequency);
                     ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:138:5: error: 'i2cSetFrequency' was not declared in this scope
     i2cSetFrequency(i2c, frequency);
     ^~~~~~~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:138:5: note: suggested alternative: 'getApbFrequency'
     i2cSetFrequency(i2c, frequency);
     ^~~~~~~~~~~~~~~
     getApbFrequency
C:/Users/techo/.platformio/lib/src/Wire.cpp: In member function 'size_t TwoWire::getClock()':
C:/Users/techo/.platformio/lib/src/Wire.cpp:143:28: error: 'i2c' was not declared in this scope
     return i2cGetFrequency(i2c);
                            ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:143:12: error: 'i2cGetFrequency' was not declared in this scope
     return i2cGetFrequency(i2c);
            ^~~~~~~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:143:12: note: suggested alternative: 'getApbFrequency'
     return i2cGetFrequency(i2c);
[... 152 lines cut ...]
C:/Users/techo/.platformio/lib/src/Wire.cpp:380:10: note: suggested alternative: 'i2cRead'
   return i2cDebug(i2c,setBits,resetBits);
          ^~~~~~~~
          i2cRead
C:/Users/techo/.platformio/lib/src/Wire.cpp: In member function 'bool TwoWire::busy()':
C:/Users/techo/.platformio/lib/src/Wire.cpp:384:25: error: 'i2c' was not declared in this scope
   return ((i2cGetStatus(i2c) & 16 )==16);
                         ^~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:384:12: error: 'i2cGetStatus' was not declared in this scope
   return ((i2cGetStatus(i2c) & 16 )==16);
            ^~~~~~~~~~~~
C:/Users/techo/.platformio/lib/src/Wire.cpp:384:12: note: suggested alternative: 'i2cGetClock'
   return ((i2cGetStatus(i2c) & 16 )==16);
            ^~~~~~~~~~~~
            i2cGetClock
*** [.pio\build\openevse_esp32-gateway-e_dev\lib359\src\Wire.cpp.o] Error 1
====================================================================================== [FAILED] Took 149.54 seconds ======================================================================================
Environment                   Status    Duration
----------------------------  --------  ------------
openevse_esp32-gateway-e_dev  FAILED    00:02:29.545
================================================================================= 1 failed, 0 succeeded in 00:02:29.545 ================================================================================= 

 *  The terminal process "C:\Users\techo\.platformio\penv\Scripts\platformio.exe 'debug'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
[image]

jeremypoulter, 2023-08-03:
That file is auto generated by a python script, is using windows `\` and breaking the build

matth-x, 2023-08-18:
Update: #682 resolves the LITTLEFS issue.

matth-x, 2023-10-23:
A further outstanding issue on Windows is the upload command in the platformio.ini (env:openevse_wifi_v1):

```
upload_command = curl -F firmware=@$SOURCE http://$UPLOAD_PORT/update --progress-bar | cat
```

This only works on Linux. Since I prefer working with the Serial flasher anyway, I don't have a fix at hand.

jeremypoulter, 2023-10-23:
> A further outstanding issue on Windows is the upload command in the platformio.ini (env:openevse_wifi_v1):
>
> ```
> upload_command = curl -F firmware=@$SOURCE http://$UPLOAD_PORT/update --progress-bar | [...]

I would just use a different board for dev work, the OpenEVSE board is not great for dev work as it only has the one serial port exposed

matth-x, 2023-10-23:
Good tip, the extra debug port would be very handy and I will change my dev setup then.

Apart from that line in the platformio.ini I did not face any further issues on Windows.

zerodur12, 2023-12-03:
I have also some issue when compiling on window. 
My objective was to test a modification on the latest OpenEVSE WiFi firmware [v5.0.2](https://github.com/OpenEVSE/openevse_esp32_firmware/releases/tag/v5.0.2) to test some modification on the [current_shaper.cpp](https://github.com/OpenEVSE/openevse_esp32_firmware/blob/master/src/current_shaper.cpp) to try to solve the shapper issue #762 by supressing line https://github.com/OpenEVSE/openevse_esp32_firmware/blob/e8219948309112c3fb3d73927b05120c058bfc60/src/current_shaper.cpp#L177C1-L179C3 re-compile and test.

But before changing any thing my first test was to verify that I was able to recompile the firmware

for that I'm working with Visual Studio Code under windows, I download https://github.com/OpenEVSE/openevse_esp32_firmware/releases/download/v5.0.2/openevse_wifi_v1.bin open the project and compile it with out any modification. 
It compile with success with some warning about GUI v2 file missing:

Processing openevse_wifi_v1 (board: esp32dev; platform: espressif32@6.0.1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Firmware Revision: -D BUILD_TAG=local___modified -D BUILD_HASH=_modified
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.20.1
|-- ArduinoMongoose @ 0.0.18
|-- Micro Debug @ 0.0.5
|-- ConfigJson @ 0.0.5
|-- OpenEVSE @ 0.0.14
|-- ESPAL @ 0.0.3
|-- StreamSpy @ 0.0.1
|-- MicroTasks @ 0.0.3
|-- ArduinoOcpp @ 0.3.0
|-- ArduinoOcppMongoose @ 0.1.0
|-- Adafruit NeoPixel @ 1.11.0
|-- Adafruit MCP9808 Library @ 1.1.2
|-- EEPROM @ 2.0.0
|-- LittleFS @ 2.0.0
|-- Update @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Ethernet @ 2.0.0
|-- FS @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
10.2.4
Warning: GUI files not found, run 'git submodule update --init' (C:\Users\zerod\OneDrive\Documents\openEvse\openevse_esp32_firmware-5.0.2\openevse_esp32_firmware-5.0.2\gui-v2)
PATH=C:\Users\zerod\.platformio\packages\toolchain-xtensa-esp32\bin;C:\Users\zerod\.platformio\packages\tool-mkspiffs;C:\Users\zerod\.platformio\packages\tool-mklittlefs;C:\Users\zerod\.platformio\packages\tool-mkfatfs;C:\Users\zerod\.platformio\packages\tool-esptoolpy@1.40400.0;C:\Users\zerod\.platformio\penv\Scripts;C:\Users\zerod\.platformio\penv;C:\Users\zerod\.platformio\python3;C:\WinAVR-20100110\bin;C:\WinAVR-20100110\utils\bin;C:\Program Files\MATLAB\R2023b\bin;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\zerod\AppData\Local\Microsoft\WindowsApps;C:\Users\zerod\AppData\Local\Programs\Python\Python38-32\Scripts\;C:\Users\zerod\AppData\Local\Programs\Python\Python38-32\;C:\Program Files (x86)\Intel\Trace Analyzer and Collector\9.0.3.050\dll;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Users\zerod\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\zerod\AppData\Roaming\npm;C:\Users\zerod\AppData\Local\GitHubDesktop\bin
Retrieving maximum program size .pio\build\openevse_wifi_v1\firmware.elf
Checking size .pio\build\openevse_wifi_v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.8% (used 61544 bytes from 327680 bytes)
Flash: [========= ]  93.9% (used 1846321 bytes from 1966080 bytes)
===================================================================== [SUCCESS] Took 22.03 seconds =====================================================================

Environment       Status    Duration
----------------  --------  ------------
openevse_wifi_v1  SUCCESS   00:00:22.034
====================================================================== 1 succeeded in 00:00:22.034 ======================================================================

and I test the generated firmware.bin file by install it on my  EmonEVSE WiFi Connected EV Charging Station (Type-2).
After the OTA Download of the firmware everething seems to be ok and the interface was nominal. But if I switch off and then switch on the EmonEVSE the EmonEVSE stop at the 'initialisation phase' on the display and don't go further

downloading back v5.0.2 openevse_wifi_v1.bin file make the EmonEVSE work again

alfredoanton82, 2023-12-06:
Hi, 

I've spend half afternoon trying to figure this LittleFS error with v2_gui tag until I saw it was an actual issue... XD, Thanks for the fix on the littleFS @matth-x!! 

Found the development guide while investigating, (@zerodur12 [link])(https://github.com/OpenEVSE/openevse_esp32_firmware/blob/master/docs/developer-guide.md) it compiles nicely after installing node.js and build the GUI first... and uploads, directly from platformIO through OTA to an ESP32.

Although it is working as I can see the LCD and the RGB led, and I can connect via web to the openevse, the GUI is simply not shown, no error either... just nothing in the screen. 

Any hint? Something I'm missing? Thanks in advance!

Thanks!
Alfredo

zerodur12, 2023-12-10:
@alfredoanton82 
Hi
I have also follow the development guide but there still some warning during compilation and if I upload the compiled firmware
and like you "I can connect via web to the openevse, the GUI is simply not shown, no error either... just nothing in the screen."

closed 2026-09-21 by github-actions[bot]