OpenEVSE/openevse_esp32_firmware #710: Session kWh just keeps goin up between different sessions?

pir8radio, 2023-08-04:
The session KWH should show only what is/was consumed during current session or plug-in.  If I unplug and then plug back in, session should zero out and start a new SESSION...   right now session just keeps incrementing regardless of how many different cars charge You can see in the first screenshot today I have charged one vehicle and that vehicle used 7kwh yet session shows 18kwh from all previous charges days ago.  [image] 

[image]

fhteagle, 2023-08-12:
Just tested against master branch dev from about a week ago, session counter definitely reset to 0 at J1772 physical removal event. Does not seem to reset session on a power cycle though. Not sure what the intended behavior of that is supposed to be.

KipK, 2023-08-12:
It is made to survive reboot yes. 

But I wonder where @pir8radio  problem comes from.
What is you Pause mode in settings/Evse? Is it set to disabled?

pir8radio, 2023-08-12:
I will close this issue.. it seems this particular charger had really old EVSE firmware.   I made a mistake,  Updating the EVSE firmware seems to have fixed my issue..   Sorry to waste your time.

closed 2023-08-12 by pir8radio