OpenEVSE/openevse_esp32_firmware #716: Local date format and its further processing

NJbubo, 2023-09-01:
The date and time showing wrong.
Local date format is "2023. 09. 01".
I  think the local date and time format and its further processing is wrong method.
It doesn't include these separator characters, and not where the program expects them.
The year number not the last element of the date.
[image]
[image]
[image]

jeremypoulter, 2023-09-01:
[image]

The year and separator are down to the localisation settings of your OS/Browser I think so that is correct, but indeed there do seem to be a few rendering issues with those settings. What Timezone / language do you have your browser/OS set to and what Browser / OS are you using?

NJbubo, 2023-09-01:
Hi,
My language and localisation settings : Hungarian. Timezone : Europe / Budapest. 
The pictures were taken in the android / chrome system. Same date in Windows 10 / Edge.

The "openevse-gui-v2/src/lib/utils.js" file 82 row: 
     const arr = d.toLocaleString(DateTime.DATETIME_SHORT).split(" ")
Your result with split (" "): arr[0] = date, arr[1] = time, arr[2] = AM or PM.
My result with split (" "): arr[0] = year, arr[1] = month, arr[2] = day, arr[3] = time, ...

jeremypoulter, 2025-11-11:
We keep the GUI issues in this repo. You can still reference the issue, you just need to add the org/repo IIRC

NJbubo, 2025-11-11:
OK. I didn't know that.

closed 2026-09-21 by github-actions[bot]