OpenEVSE/openevse_esp32_firmware #718: Monitoring: Voltage with too many decimals

dml33, 2023-09-05:
Hi.

I am running v4.2.2:

[image]

and I am noticing that there is a small issue with the voltage shown in the Monitoring section.

I am sending via MQTT the `current_voltage` topic with just one decimal (228.61). But, OpenEVSE is recalculating it with 8 decimals (MQTT topic `voltage`: 228.6000061) and it is also shown with 8 decimals (228.6000061) in the GUI v2:

[image]

Maybe there is small issue there.

Thanks.

dml33, 2023-09-30:
Hi.

It is still happening in v5.0.0

[image]

[image]

Thanks

KristianDebrenti, 2023-10-01:
I confirm.
[image]

dml33, 2023-11-03:
I can confirm that this issue still happens with v5.0.2 firmware.

[image]

closed 2026-09-21 by github-actions[bot]