OpenEVSE/openevse_esp32_firmware #724: several network functions not working since upgrade to 5.0.0

ve6rah, 2023-09-20:
I upgraded to 5.0.0 this morning, and ever since that time MQTT and NTP aren't working. All the settings show the same in the front-end as they did before the update, and I have double checked that both MQTT and NTP servers are working properly.

Unsure how to troubleshoot, I don't see much in the way of logs, in developer debug console I see:
`{"mqtt_connected":0}`

Ideas?

ve6rah, 2023-09-20:
After a LOT of digging, and no thanks at all to OpenEVSE's lack of logging/troubleshooting ability, I discovered that it seems to be a DNS issue. I have no idea WHY it's a DNS issue, but when I changed NTP and MQTT servers to IP addresses instead of names, it works.

Still no idea how to troubleshoot the DNS issue though, everything else connected to the same wifi has no problem with the DNS.

jeremypoulter, 2023-09-20:
That sounds really weird, can't think of anything that has changed in that area.

Yeah the debug in the production builds is not great, we do have a ticket to improve that.

ve6rah, 2023-09-20:
I tried multiple reboots, I tried a factory reset, no luck. It's obviously having DNS issues, but I don't know where to go from here.

ve6rah, 2023-09-25:
DNS servers are local to my network. Or at least that is what the DHCP server is serving, without any form of diagnostics, I can't see if OpenEVSE is trying to access a different DNS service, if that is the case, it would be unsuccessful, as I have firewall off my IoT devices from the public internet.
However it would also be poor form for it to ignore the DNS servers supplied through DHCP.

matth-x, 2023-09-26:
The networking library (mongoose) uses the Google DNS server by default (8.8.8.8), but unfortunately, it doesn't try alternative DNS servers automatically. It should be possible to change the default by inserting

`-D MG_DEFAULT_NAMESERVER='"8.8.8.8"'` (with the alternative IP address)

into the build_flags section of the platformio.ini ([here](https://github.com/OpenEVSE/openevse_esp32_firmware/blob/ceef3d13931df12c0c91a6348b2beaf51f77bfb6/platformio.ini#L71)).

fhteagle, 2023-09-26:
@ve6rah -

If you have a setting on your router to redirect all DNS requests to local DNS source, that would be a possible workaround for now. I have this set up on my custom LAN router that uses dnsmasq for both DNS and DHCP, so "misbehaving" devices still end up using local DNS whether they want to or not.

Long term fix should be having OpenEVSE respect DNS servers assigned by DHCP, of course.

ve6rah, 2023-09-26:
Right now my short-term fix is to ignore DNS, and put everything in by IP.

However it seems like we've at least figured out why this is happening. Sounds like a badly misbehaving library.

jeremypoulter, 2023-09-26:
Hum, my wrapper does try to select the router provided IP via `MongooseCore::ipConfigChanged()` maybe that is not being called...

jeremypoulter, 2023-09-26:
Also what DNS names are you trying to resolve?

ve6rah, 2023-09-26:
The names I'm trying to resolve are both internal to my network. One is the MQTT server, the other is the NTP server. I don't allow IoT devices to connect to the public internet.

ve6rah, 2023-09-26:
The domain names in question are FQDN.
They are part of my registered domain, however only served locally.

ve6rah, 2026-09-21:
Seems like it's only partially solved. On the newest firmware it's accepting the primary DNS, but not the secondary, so if the primary goes down, it has no DNS (there's a reason internet standards ask for a minimum of 2 DNS servers)

RAR, 2026-09-21:
Reopening — the follow-up report is a real, reproducible limitation rather than inactivity.

**Root cause:** the firmware's networking library (ArduinoMongoose) runs its own resolver with a **single** nameserver. `MongooseCore::ipConfigChanged()` walks both DHCP DNS slots but stops at the first valid IPv4 entry and hands only that one to `mg_set_nameserver()`; the second slot is never stored. When that primary stops answering, `resolve_cb()` in `mongoose.c` gets `MG_RESOLVE_TIMEOUT` and fails the connection outright — there is no retry against the other server. Everything that resolves through Mongoose is affected: MQTT, OCPP, the HTTP client (Emoncms, Home Assistant, OTA/update checks).

NTP is not in that set: `time_man.cpp` uses `getaddrinfo()`, which goes through LwIP's resolver, and LwIP does fail over to the second DHCP server. So on current firmware you should see NTP keep working when the primary goes down while MQTT stalls — that split is the signature of this bug.

**Fix shape:** keep both DHCP-supplied servers in `MongooseCore`, and on a resolve timeout re-issue the query once against the secondary before failing. The timeout path in `resolve_cb()` is already locally patched (the IPv6 AAAA→A fallback lives there), so the retry slots in beside it. The Mongoose 7 line that #1155 moves to has the same single-`dns4` model, so the change is needed there too; a PR against ArduinoMongoose can follow.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-10-11 by jeremypoulter