OpenEVSE/openevse_esp32_firmware #727: Self production Export not working

xtreamjava, 2023-09-23:
I have mqtt topic providing +/- values in w. After new firmware update, it stopped working when I set to export.
[image]

But when I switch to production it starts displaying values
[image]

xtreamjava, 2023-09-23:
Fixed after reset.

closed 2023-09-23 by xtreamjava

danepowell, 2023-10-05:
Seeing this issue clued me into a related issue: the upgrade from v4.2.2 to v5.0.0 doesn't preserve the production vs export setting. Mine was previously set to export and somehow got switched to production after the upgrade. I wouldn't be surprised if this is related to you having the "jiggle the handle" (reset) to get it working.

Zuikkis, 2024-05-08:
I have this very same issue and it's not fixing itself.. :(

I upgraded from 4.x to 5.1. In Export mode, I can't see negative watts at all, if mqtt sends -5000 watts in OpenEVSE is showing 0W. Positive watts do show.

Zuikkis, 2024-05-08:
Ok, factory reset fixed it. But apparently the settings were ruined by firmware upgrade 4->5.