OpenEVSE/openevse_esp32_firmware #735: Unable to cancel manual override using [x] button

glynhudson, 2023-09-27:
There used to be a little [x] click-box next to 'manul override', but this seems to be missing in the latest master version 

[image]

This is what it used to look like, but it was normally white:

[image]

KristianDebrenti, 2023-10-10:
@glynhudson also, are you able to adjust the slider of the Charge Rate? I am unable to do so.

glynhudson, 2023-10-10:
Yes, current adjustment works fine

closed 2026-09-21 by github-actions[bot]