OpenEVSE/openevse_esp32_firmware #753: Change Default State "Disabled" to "Active"

chris1howell, 2023-10-27:
The un-configured and default value for default state is "Disabled" in firmware version v5.0.0. 

This is causing every station/kit shipped with 5.0.0 to Sleep and not operate right out of the box. Users who upgrade to 5.0.0 on a working station are left with a station that no longer operates as it did previously. Additionally, users who do not understand the meaning of the Default State value and accept the default value "Disabled" in the Wizard are left with a station that does not operate. Many users are contacting support or downgrading firmware. 

The un-configured state should be changed to "Active". The default option should be "Active", the user should have to purposefully change the setting from "Active" to "Disabled".

Out of the box stations and firmware upgrades should work by default and not require the user to change settings to have a functioning station.

jeremypoulter, 2024-01-09:
This is a little tricky as these bool config vars are just part of the flags and we don't really know if they have been explicitly set by the user. We probably need to add a mask of flags that have changed so we can do this.

TechnicalLee, 2024-01-23:
Also had this issue after updating to 5.0.2 until I figured out the default setting disallowed charging. I definitely agree this should be changed to enabled by default so it doesn’t lock out EVSEs after firmware updates.

jeremypoulter, 2024-01-23:
I have a fix for this in #796, just need to do some testing around it

PR #796 (2024-01-29): Set the default mode to active
This adds a change to allow the default flags to be updated. Only the flags explicitly saved by the user are loaded. This may loose the current configured default state, but it is causing confusion for many users.

fixes #753
closed 2024-01-29 by glynhudson