OpenEVSE/openevse_esp32_firmware #755: Blank page after compiling current version

jdgarcia99, 2023-10-29:
Hi

I'm compiling master version 4.2.2 and uploading the project with a ESP32 WROOM DevKit C.

I did all steps as the documentation says:

1. First build the web assets (for v2)

```
git submodule update --init --recursive
cd gui-v2
npm install
```

Then

`ppm run build`

After this I compile with Platformio in VC Studio the firmware without any problem

Upload to the ESP32 goes well

Then the ESP32 start in AP mode as normal, connect to their wifi and when I access to ip 192.168.4.1, a BLANK Page appears.

Tried everything but no success

If I try with the status page through http://192.168.4.1/status this is what I get

[image]

Any help would be appreciated

Thanks

jeremypoulter, 2023-10-29:
The latest version is 5.0.0 are you sure you have an up-to-date source tree?

jdgarcia99, 2023-10-29:
> The latest version is 5.0.0 are you sure you have an up-to-date source tree?

In fact I'm using this repository cloned directly from this link)

https://github.com/OpenEVSE/ESP32_WiFi_V4.x.git

jdgarcia99, 2023-10-29:
I'm trying to create the wt32-eth01 board version for the stable release 4.2.2 with gui-v2. Could you do that and publish with the rest of binaries?

wt32-eth01 is from 5.0.0 onwards which is different

Thanks

Mawverick, 2024-02-16:
Experience the same issue  (blank page) while compiling GUI V2 with any firmware. (env:adafruit-huzzah32) The GUI works fine once wifi configuration done and ip assigned. Moreover the GUI works fine in AP mode using hostname openevse-xxxx.local but not via AP i adress 192.168.4.1.
Issue described below fixed in a code as well so shall not be the case.
https://github.com/OpenEVSE/openevse_esp32_firmware/issues/791

closed 2026-09-21 by github-actions[bot]