OpenEVSE/openevse_esp32_firmware #760: Wrong Monitoring statistics after updating

dml33, 2023-11-03:
Hi.

When updating OpenEVSE Wifi to latest 5.0.2, I notice that `Year`, `Month` and `Week` statistics in the monitoring section become the ones which were the `Week` ones before the updating.

[image]

Thus, `Year` and `Month` previous statistics are lost.

Firstly, I noticed it when I updated from 4.2.2 to 5.0.0, but I can confirm that it also happens when updating from 5.0.0 to 5.0.2.

Thanks

Mathopenevse, 2023-11-09:
Hi, Do you tried to save and import firmware backup before?

dml33, 2023-11-10:
Hi.

I did not do anything about backups.

I just updated from the Firmware section and I chose direct install.

[image]

[image]

TommySharpNZ, 2024-02-21:
I just upgraded today and notice similar issues....
In my case the Today value is incorrect and is even less than the current session... But I guess that will fix itself tomorrow :-)

But the Year value is the same as the Month value...

Is there any way to fix up the Year value if I know what it should be?

[image]

closed 2026-09-21 by github-actions[bot]